ARMmbed/mbed-os-example-ble #23: examples are not alligned to API of the mbed-evets library

nvlsianpu, 2016-08-17:
Some changes were made in mbed-events etc: https://github.com/ARMmbed/mbed-events/pull/7, https://github.com/ARMmbed/mbed-events/pull/5. Source of examples (main.cpp) were not update. Therefore examples even do not build. Is someone trying to resolve this? If not - I can do that.

pan-, 2016-08-17:
@nvlsianpu Which examples does not build ? 
We are using mbed-events#v0.4 in our examples and the changes you pointed are not in this version.

nvlsianpu, 2016-08-17:
All examples. Yes I used this version and lastest - mentioned changes were already in v4.0.

nvlsianpu, 2016-08-17:
I checked - Examples (at last tag mbed-os-5.1.0 using mbed-events#v0.3. This works well for GCC_ARM, but neither ARM5 nor IAR do not build.

nvlsianpu, 2016-08-17:
I checked - Examples (at last tag mbed-os-5.1.0 using mbed-events#v0.3. This works well.

closed 2016-08-17 by nvlsianpu