ARMmbed/mbed-os-example-ble #25: nRF51-DK printf() causes hang when PC serial port not active / opened

nuket, 2016-08-23:
Already reported this once in the [mbed OS codebase](https://github.com/ARMmbed/mbed-os/issues/2489), here's another concrete example of this happening. 

The problem is that, with the nRF51-DK hooked up to my computer, if I don't connect to and open the serial port exposed as `/dev/ttyACM0`, whatever software is running on the DK board will just hang. It seems like the Nordic serial port code locks up because it *always* expects some hardware flow control of some sort, so that if I don't have `minicom` running, it doesn't work. Also, if I run `minicom` then exit from it, the DK board will keep running for a while, then eventually hangs.

The Nordic serial port code simply does not work properly. Where, in the HAL (pun intended), do I fix that?

The video shows the code example from https://github.com/nuket/mbed-os-example-ble/tree/nordic-serial-bug/BLE_LEDBlinker, which is running:
1. At the start, the board is hung, because I don't have `minicom` open. 
2. I open `minicom`, the board starts printing "Alive." again.
3. I close `minicom`, the board eventually hangs again.

nuket, 2016-08-23:
Here's the `.mbed` configuration I'm using:

> $ cat .mbed
> TARGET=NRF51_DK
> TOOLCHAIN=GCC_ARM
> GCC_ARM_PATH=/home/max/devtools/gcc-arm-none-eabi-5_4-2016q2/bin
> ROOT=.

I've tried with GCC 4.9 and 5.4, both exhibit identical behavior.

nuket, 2016-08-23:
Related issue, possibly: https://github.com/ARMmbed/mbed-os/issues/2474

sg-, 2016-08-25:
Can you confirm is #2535 resolved this issue?

nuket, 2016-08-26:
I will take a look.

nuket, 2016-08-30:
Confirmed https://github.com/ARMmbed/mbed-os/pull/2535 resolved the issue after adding to `mbed_app.json`:

```
"NRF51_DK": {
    "target.uart_hwfc": 0
}
```

closed 2016-08-30 by nuket