ARMmbed/mbed-os-example-ble #27: BLE_BUTTON_PIN_NAME is not defined

janjongboom, 2016-08-29:
In main.cpp for BLE_Button we use `BLE_BUTTON_PIN_NAME`. I see in mbed_app.json that we have something there, but I don't think the macro name is correct. 

https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_Button/mbed_app.json

pan-, 2016-08-31:
@janjongboom What is your issue ? Could you provide me more details ?

janjongboom, 2016-08-31:
@pan- 

If you build BLE_Button with mbed-cli it works fine, macro is there.

If you build BLE_Button in online compiler the macro is not defined.

pan-, 2016-08-31:
Thanks for the clarification. 

Did you tried it from the [example](https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-ble-Button/) on developer.mbed.org ?
It works for me.

janjongboom, 2016-08-31:
@pan- Ah, didn't realize we had them split out there. Yeah I assume that will work, given that it's not in a different folder now. I'll share that link in the future.

closed 2016-08-31 by janjongboom