ARMmbed/mbed-os-example-ble #35: OOB: BLE_EddystoneService android application not working as in instructions

mikaleppanen, 2016-12-09:
Installed "Physical Web" application from google apps. Current version is 0.2.1. Looks like it migth not support configuration edit interface anymore or there is some problem that since it does not appear. It also crashed several times on my Samsung J5 when making the scan when mbed device was on. After the crashes it sometimes displayed the mbed OS URL.

pan-, 2016-12-09:
The version of the application used in the instructions is the version 0.1.865. 
I can update them to reflect the changes made to the application (`Remove beacon configuration`); will you be happy with such change ? 
For the crash of the application; I'm afraid there is nothing which can be done on the beacon side.

mikaleppanen, 2016-12-09:
Yes, if the configuration is no longer possible with the current application.

pan-, 2016-12-12:
@mikaleppanen Could we close this ? 
I've merged #36 .

mikaleppanen, 2016-12-12:
Yes, it is ok to close this.

closed 2016-12-12 by pan-