ARMmbed/mbed-os-example-ble #37: OOB: BLE_EddystoneService beacon url configuration cannot be changed in beetle board

mikaleppanen, 2016-12-11:
Beacon URL cannot be changed with Beetle with Android "nRF Master Control Panel" . With NUCLEO-F401RE and nRF-51dk URL does change.

With working boards "nRF Master Control Panel" configuration interface "Data" sections shows "Value: https:://www.mbed.com" or whatever is configured. 

With Beetle when "Data" is read is shows empty value field "Value: "

Also in programs main display URL remains as mbed.com and is not changed to arm.com.

Happend with both online compiler and when compiling locally with ARM compliler.

pan-, 2016-12-12:
@fvincenzo Do you know what is going on with this ? 
Is it related to the length of the characteristic ?

fvincenzo, 2016-12-12:
Adding @abhishek-pandit @mmorenobarm to follow up.

closed 2017-11-03 by pan-