ARMmbed/mbed-os-example-ble #39: [NRF5] Impossible to debug or flash the examples with IAR

pan-, 2016-12-12:
* **Description**: It is not possible to flash or debug an application with IAR on the Nordic target. This issue is caused by a bad interaction between CMSIS-DAP IF present on the Nordic boards and IAR CMSIS-DAP debug driver.   
* **Workaround**: The best solution is to switch to the JLINK IF image and configure the IAR project to use it.
    - Download the [JLink IF](https://www.nordicsemi.com/eng/nordic/download_resource/52276/5/42250904) and install it on the board.
    - Open the exported project in IAR then go to its options. 
    - Go to the debugger section 
        + In the setup tab choose J-link/J-trace as the debug driver.
        + In the Download tab disable the `Use flash loader(s)` option.
        + In the Extra options tabs, tick the `Use command line options` and add the following content: `--drv_vector_table_base=0x0`.

0xc0170, 2017-03-07:
One question - what's the IAR version for this known issue ? Is this still valid for latest IAR stable version - 7.8 ?

pan-, 2017-03-07:
yes, it is still valid for the latest stable version of IAR. 
The issue lie in the CMSIS DAP IF used on Nordic boards.

edablan, 2018-08-01:
Is this valid for the UBLOX EVK-NINA-B1? It does not use the CMSIS-DAP, but uses a built in J-Link debugger. Also noticing issues in debug.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

pan-, 2020-10-02:
I'm closing this, IAR isn't supported by Mbed OS 6.

closed 2020-10-02 by pan-