ARMmbed/mbed-os-example-ble #41: OOB: BLE_Beacon example with make_gcc_arm has link error (using NRF52_DK)

sunsmilearm, 2016-12-12:
command: >>mbed export -m NRF52_DK -i make_gcc_arm
                  >>make

Error message:

```
"link: BLE_Beacon.elf"
'arm-none-eabi-objcopy' -O binary BLE_Beacon.elf BLE_Beacon.bin
'arm-none-eabi-objcopy' -O ihex BLE_Beacon.elf BLE_Beacon.hex
"NOTE: the srec_cat binary is required to be present in your PATH. Please see http://srecord.sourceforge.net/ for more information."
srec_cat .././mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132/hex/s132_nrf52_2.0.0_softdevice.hex  -intel BLE_Beacon.hex -intel -o BLE_Beacon-comb
ined.hex -intel --line-length=44
process_begin: CreateProcess(NULL, srec_cat .././mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132/hex/s132_nrf52_2.0.0_softdevice.hex -intel BLE_Be
acon.hex -intel -o BLE_Beacon-combined.hex -intel --line-length=44, ...) failed.
make (e=2): The system cannot find the file specified.
make[1]: *** [BLE_Beacon-combined.hex] Error 2
make: *** [all] Error 2
```

pan-, 2016-12-12:
> "NOTE: the srec_cat binary is required to be present in your PATH. Please see http://srecord.sourceforge.net/ for more information."
srec_cat .././mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132/hex/s132_nrf52_2.0.0_softdevice.hex  -intel BLE_Beacon.hex -intel -o BLE_Beacon-comb
ined.hex -intel --line-length=44

Did you download and install srecord as indicated in the `NOTE` message ?

sunsmilearm, 2016-12-13:
@pan- Passed after installed that tool. Thank you.

closed 2016-12-14 by pan-