ARMmbed/mbed-os-example-ble #45: The mbedtls/* definition in file .mbedignore causes a compile error

ccchang12, 2017-01-11:
I compile BLE_BatteryLevel example for target NUMAKER_PFM_NUC472 but get a error.

The root cause is the NUC472 supports H/W SHA and need to include mbedtls header. Due to the mbedtls/* definition in file .mbedignore, the H/W SHA code cannot find the header and encounters an error. Please remove the mbedtls/* definition to fix this issue.

Below is the compile error message.

......
Scan: mbed
Scan: env
Compile [ 73.6%]: sha1_alt.c
[ERROR] .\mbed-os\targets\TARGET_NUVOTON\TARGET_NUC472\crypto\sha\sha1_alt.c:21:28: fatal error: mbedtls/config.h: No such file or directory
compilation terminated.
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\cczhang\mbed\mbed-os-example-ble\BLE_BatteryLevel\mbed-os\tools\make.py -t GCC_ARM -m NUMAKER_PFM_NUC472 --source . --build .\BUILD\NUMAKER_PFM_NUC472\GCC_ARM" in "C:\Users\cczhang\mbed\mbed-os-example-ble\BLE_BatteryLevel"

0xc0170, 2018-01-05:
@pan- @paul-szczepanek-arm I referenced this in mbed OS issue, this got our attention this week as these examples cant be build if a component or target (as reported above) includes mbedtls that is part of mbed OS. 

I can't find the reason in the git log (almost no commit message ).

pan-, 2018-01-08:
@0xc0170 The .mbedignore file is present for historical reasons, we can remove it. However it doesn't sounds healthy that the target HAL depends on a feature.

0xc0170, 2018-01-08:
The latest master - nuvoton have crypto in mbedtls folder, not as previously was in target (see https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NUVOTON/TARGET_NUC472). 

>  The .mbedignore file is present for historical reasons, we can remove it. 

This should fix this issue! Thanks

donatieng, 2018-06-11:
Closing this issue as this would have been resolved in mbed OS 5.8.

closed 2018-06-11 by donatieng