ARMmbed/mbed-os-example-ble #47: Error: L6406E: No space in execution regions with .ANY selector matching ...

x893, 2017-01-30:
Hi,
i compile example from BLE_Beacon example for HRM1017 board under mbed-cli and online compiler.

But only this messages on link
Link: BLE_Beacon
Error: L6406E: No space in execution regions with .ANY selector matching libspace.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching retarget.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching stdio_streams.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching stdio_streams.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching stdio_streams.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching btle.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching ble_conn_params.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching rt_Task.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching equeue_mbed.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching pstorage.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching rt_List.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching pstorage.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching RTX_Conf_CM.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching FileBase.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching BLE.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching RTX_Conf_CM.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching us_ticker.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching pwmout_api.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching softdevice_handler.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching RTX_Conf_CM.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching serial_api.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching device_manager_peripheral.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching fds.o(fs_data).
Error: L6406E: No space in execution regions with .ANY selector matching RTX_Conf_CM.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching RTX_Conf_CM.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching rt_CMSIS.o(.bss).
Error: L6406E: No space in execution regions with .ANY selector matching rt_CMSIS.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching BLE.o(.data).
Error: L6406E: No space in execution regions with .ANY selector matching nordic_critical.o(.data).
...

Can you help to resolve this issue ?

Thanks

pan-, 2017-01-30:
@x893 There is not enough RAM on your target to support this example out of the box. 

You can tweak the application to disable components like the RTOS to make it fit on your target. To do that, create a `.mbedignore` file at the root of the project and paste the following content: 

```
mbed-os/rtos/*
mbed-os/features/FEATURE_CLIENT/*
mbed-os/features/FEATURE_COMMON_PAL/*
mbed-os/features/FEATURE_UVISOR/*
mbed-os/features/frameworks/*
mbed-os/features/net/*
mbed-os/features/netsocket/*
mbed-os/features/storage/*
```

Other tweaks can be applied in `mbed_app.json` (in the project root folder): 

```json
{
    "macros": [
        "NDEBUG=1",
    ], 
    "target_overrides": {
            "*": {
                    "platform.stdio-flush-at-exit": false
            },
        "K64F": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["ST_BLUENRG"],
            "target.macros_add": ["IDB0XA1_D13_PATCH"]
        },
        "NUCLEO_F401RE": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["ST_BLUENRG"]
        }
    }
}
```

Unlike the original file, `NDEBUG` is defined and standard IO will not be flushed at exit. 

On a final note, the solutions exposed here are just workarounds which will save few bytes of RAM they might not be enough depending on the application.

x893, 2017-01-30:
Thanks ! Now compile. How you know which modules can remove ? Telepathy ?

pan-, 2017-01-31:
@x893 During the transition to mbed OS 5 this issue arised with targets with less than 16K of RAM like yours. For this specific reason they do **not** officially support mbed OS 5 (mbed OS 2 is still supported).
Ignoring the RTOS brings the project to a state very close to mbed OS 2 and therefore allows the project to compile and  run.

Please, close this issue if you are happy with the answer.

closed 2017-01-31 by x893

Blue7089, 2018-03-21:
Can please tell me briefly where and how to create this file. I am completely beginner.Thanks in advance