ARMmbed/mbed-os-example-ble #49: Incorrect event size in eventQueue definitions

nuket, 2017-02-05:
The eventQueue definition in the various BLE examples is too small:

```
static EventQueue eventQueue(
    /* event count */ 10 * /* event size */ 32
);
```

I believe the event size should use the EVENTS_EVENT_SIZE define (which is currently 52 bytes).

pan-, 2017-02-23:
> The eventQueue definition in the various BLE examples is too small: 

What issue did you encounter ? 

> Shouldn't the event size should use the EVENTS_EVENT_SIZE define (which is currently 52 bytes)?

Previously they were no define for the event size and it has grown. 
I agree that it is not consistent and the define should be use.

nuket, 2017-02-23:
I didn't experience any issues due to this, just something I noticed which may cause a surprise at some point for someone starting out with the examples.

closed 2017-06-13 by pan-