ARMmbed/mbed-os-example-ble #53: Thermometer change Notify to Indicate?

jmjobin, 2017-02-16:
On the BLE_Thermometer: cannot find where it set the Notify, but it is Notify. If you want to use the nRF ToolBox we need an "Indicate" to follow the thermometer BLE specification. Do not find a clue on mbed where to change this and how to call the class to set this Indicate   :-( 
Can somebody help me?

jmjobin, 2017-02-22:
Found the solution: The Characteristic is set to Notify in the "HealthThermometerService.h". I do not know how to access the original file so I recreate mine using a HealthThermometerService.h file found on the NET and replaced NOTIFY by INDICATE and #include my HealthThermometerService.h file.
If there is a better way, someone let me know.

donatieng, 2018-06-14:
Thanks for pointing this out @jmjobin, feel free to open a pull request with the change

LDong-Arm, 2019-05-23:
Many thanks @jmjobin, you are totally correct.

The GATT specification for Health Thermometer Service requires "indicate", see https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=238688

And I confirm the HTM applet of nRF Toolbox does require "indicate" to work.

jmjobin, 2019-05-23:
Thank you.
Have a nice day.
jean-marc
---- Messaggio originale ----

Data : 23/05/2019 - 10:40 (CEST)
A : mbed-os-example-ble@noreply.github.com
Cc : jmjobin@bluewin.ch, mention@noreply.github.com

LDong-Arm, 2019-05-24:
The fix is merged to mbed-os: https://github.com/ARMmbed/mbed-os/commit/ce775d58e9f2c3e6e50aa84778ccc16b0d989b8a

closed 2019-05-24 by 0xc0170