ARMmbed/mbed-os-example-ble #56: [OOB_5_4_RC1] link errors in exported uvision5 project

toyowata, 2017-03-01:
When exported uvision5 project is compiled, I got following link errors.

```
linking...
.\BUILD\BLE_EddystoneService.axf: Error: L6366E: analogin.o attributes are not compatible with the provided attributes .
Object analogin.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)

(snip)

Not enough information to list image symbols.
Not enough information to list the image map.
Finished: 22982 information, 0 warning and 5745 error messages.
".\BUILD\BLE_EddystoneService.axf" - 5745 Error(s), 75 Warning(s).
Target not created.
Build Time Elapsed:  00:01:05

```
Full log output is here.  
[mbed-os-example-ble_error_log.txt](https://github.com/ARMmbed/mbed-os-example-ble/files/810076/mbed-os-example-ble_error_log.txt)

## How to reproduce

```
$ git clone -b oob_test_mbed-os-5.4 https://github.com/ARMmbed/mbed-os-example-ble
$ cd mbed-os-example-ble\BLE_EddystoneService
$ mbed deploy
$ mbed export -m NRF52_DK -i uvision5
```

Open ```BLE_EddystoneService.uvprojx``` and build the project.

## Conditions

* MDK-ARM version : uVision V5.22.0.0
* Pack : ```NordicSemiconductor.nRF_DeviceFamilyPack.8.3.2```
* Device : nRF52832_xxAA

toyowata, 2017-03-01:
Related to https://github.com/ARMmbed/mbed-os-example-wifi/issues/19 and https://github.com/ARMmbed/mbed-os/issues/3807

0xc0170, 2017-03-02:
will be fixed (fix referenced above)

closed 2017-03-10 by MarceloSalazar