ARMmbed/mbed-os-example-ble #58: [OOB_5_4_RC1] Compile error in BLE_URIBeacon for all toolchain

toyowata, 2017-03-01:
## How to reproduce

```
$ git clone -b oob_test_mbed-os-5.4 https://github.com/ARMmbed/mbed-os-example-ble
$ cd mbed-os-example-ble\BLE_URIBeacon
$ mbed deploy
$ mbed compile -m NRF52_DK -t ARM
```

```
(snip)
[Error] main.cpp@96,0:  #20: identifier "DFUService" is undefined
[Error] main.cpp@96,0:  #20: identifier "dfu" is undefined
[Error] main.cpp@96,0:  #79: expected a type specifier
[ERROR] "./mbed-os/platform/toolchain.h", line 23: Warning:  #1215-D: #warning directive: toolchain.h has been replaced by mbed_toolchain.h, please update to mbed_toolchain.h [since mbed-os-5.3]
".\source\main.cpp", line 96: Error:  #20: identifier "DFUService" is undefined
".\source\main.cpp", line 96: Error:  #20: identifier "dfu" is undefined
".\source\main.cpp", line 96: Error:  #79: expected a type specifier
.\source\main.cpp: 1 warning, 3 errors

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\toywat01\Documents\mbed\oob\mbed-os-example-ble\BLE_URIBeacon\mbed-os\tools\make.py -t ARM -m nrf52_dk --source . --build .\BUILD\nrf52_dk\ARM" in "C:\Users\toywat01\Documents\mbed\oob\mbed-os-example-ble\BLE_URIBeacon"

```

mbed ls

```
BLE_URIBeacon (None)
|- mbed-os (3a27568a505b)
`- shields\TARGET_ST_BLUENRG (a5d5d0b55839)
```

I got same compile error by IAR and GCC_ARM toolchain.

pan-, 2017-03-01:
@toyowata This is a bug in the example, I will push a fix shortly.

pan-, 2017-03-02:
@toyowata If you are happy with the fix, could you close this issue ?

toyowata, 2017-03-03:
@pan- thanks.

PR #59 (2017-03-02): Fix#58: Inprove portability of BLE_URIBeacon
This set of patch improve the portability of BLE_URIBeacon example by removing references to the DFU and providing an in memory implementation of persistence functions.

It should fix #58
closed 2017-03-03 by toyowata