ARMmbed/mbed-os-example-ble #60: [OOB_5_4_RC1] NUCLEO_F207ZG Pinmap not found

mbartling, 2017-03-02:
Hi all,

working on the 5.4 oob for mbed-os-example-ble/BLE_URIBeacon on the Nucleo f207zg board and I am getting "pinmap not found for peripheral" when I run the compiled code.

ashok-rao, 2017-03-06:
@mbartling  .. can you try the instructions under "Using ST Nucleo shield on other targets" section of the docs?
@pan- .. any other suggestions?

pan-, 2017-03-06:
@ashok-rao If `IDB0XA1_D13_PATCH` is defined in `mbed_app.json` as indicated in the instructions then it compiles just fine.

mbartling, 2017-03-06:
It compiles fine with or without the patch. I get the error during runtime

mbartling, 2017-03-06:
It compiles fine with the patch, but the Physical Web app always fails to find the beacon

ashok-rao, 2017-03-06:
@mbartling , yes. This is now a known issue. Have a look at this: [https://github.com/ARMmbed/mbed-os-example-ble/issues/62]()

@sg- @pan-

mbartling, 2017-03-06:
FYI: so far it doesnt look like a pin issue. I tried the swapping the resistors on the ble shield and following the instructions on this link https://developer.mbed.org/teams/ST/code/X_NUCLEO_IDB0XA1/ Still doesnt work

pan-, 2017-03-06:
@apalmieriGH what can be wrong with the NUCLEO_F207ZG ?

apalmieriGH, 2017-03-07:
@mbartling what is the BLE shield you are using? Can you try another example to check if the same issue occurs?

pan-, 2017-03-07:
@apalmieriGH I believe he's using the` IDB05A1`, I wonder if the pin mapping of the `NUCLEO_F207ZG` is well defined.

apalmieriGH, 2017-03-07:
@mbartling can you try the suggestion by @mray19027 reported here: https://github.com/ARMmbed/mbed-os/issues/3541 ?

@pan-

MarceloSalazar, 2017-03-10:
@mbartling could you please provide an update on this?

mbartling, 2017-03-10:
Hey @MarceloSalazar, worked with @mray19027 he is having issues compiling and I am still having the pinmap not found when I run the code.

MarceloSalazar, 2017-04-05:
> Also tried to update the repository and now it no longer exists as of 5.4

Correct, BLE_URIBeacon example has been deprecated - we should use an alternative example (feel free to close this ticket an open a new one).

> he is having issues compiling and I am still having the pinmap not found when I run the code.

If this is not working in 5.4.x, could you share more information? (build logs, serial output, environment info, etc).

pan-, 2017-04-05:
@MarceloSalazar Others reported the issue [here](https://github.com/ARMmbed/mbed-os-example-ble/issues/40#issuecomment-291452560). The issue is impacting NUCLEO_F4xx based targets when they used mbed OS 5.4, the combination K64F + BLE shield is fine though.

closed 2017-06-13 by pan-