ARMmbed/mbed-os-example-ble #61: [OoB_RC1_5.4] : FLASH size exceeds when using make_gcc_arm on nRF51_DK.

ashok-rao, 2017-03-03:
Steps to reproduce:
1. Import the example, switch to OoB branch.
2. mbed export -i make_gcc_arm -m nRF51_DK
3. make clean
4. make.

Compiles fine, but at link time, fails due to FLASH overflow.
```
C:\Ashok\mbed_CLI\v5_4_OoB\mbed-os-example-ble\BLE_EddystoneService>mbed export -i make_gcc_arm -m nRF51_DK
c:\python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
<snip>

C:\Ashok\mbed_CLI\v5_4_OoB\mbed-os-example-ble\BLE_EddystoneService>make clean
rmdir /S /Q "BUILD"

C:\Ashok\mbed_CLI\v5_4_OoB\mbed-os-example-ble\BLE_EddystoneService>make
<snip>
"Compile: main.cpp"
In file included from .././mbed-os/platform/platform.h:28:0,
                 from .././mbed-os/features/filesystem/mbed_filesystem.h:23,
                 from .././mbed-os/mbed.h:51,
                 from ../source/main.cpp:19:
.././mbed-os/platform/toolchain.h:23:2: warning: #warning toolchain.h has been replaced by mbed_toolchain.h, please update to mbed_toolchain.h [since mbed-os-5.3] [-Wcpp]
 #warning toolchain.h has been replaced by mbed_toolchain.h, please update to mbed_toolchain.h [since mbed-os-5.3]
  ^
"link: BLE_EddystoneService.elf"
c:/program files (x86)/gnu tools arm embedded/5.4 2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: BLE_EddystoneService.elf section `.text' will not fit in region `FLASH'
c:/program files (x86)/gnu tools arm embedded/5.4 2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 17088 bytes
collect2.exe: error: ld returned 1 exit status
make[1]: *** [BLE_EddystoneService.elf] Error 1
make: *** [all] Error 2
```

ashok-rao, 2017-03-06:
@pan- .. any ideas? Thanks.

pan-, 2017-03-06:
@ashok-rao This is a bug in the exporters, I've opened an issue [here](https://github.com/ARMmbed/mbed-os/issues/3890)

pan-, 2017-03-10:
@ashok-rao The exporter `make_gcc_arm` use, by design, the `debug` profile which disable compilation optimization while `mbed-cli` use the `develop` profile which have compilation optimizations enabled. 

The workaround is to use the `develop` profile when the project is exported: 

```
mbed export -i make_gcc_arm -m nRF51_DK --profile develop
```

ashok-rao, 2017-03-10:
Thanks @pan- ..

closed 2017-06-13 by pan-