ARMmbed/mbed-os-example-ble #62: [OoB_RC1_5.4]: BLEURIBeacon fails to advertise itself. Cannot be detected on Physical web app.

ashok-rao, 2017-03-03:
Steps to reproduce:
1. Clone the example, update to mbed OoB branch. Install Android app. (Using version 0.2.1 - updated from Google)
2. mbed compile -t ARM -m  nRF51_DK
3. Flash binary, scan for beacons with the app.

Result:
The target enters config mode and never exits.

Detailed description:
Using a generic BLE scanner app, the beacon is first detected in config mode (after reset). After 60 seconds or more, the beacon still is detected in config mode. The Physical web app does not detect any nearby beacons.

ashok-rao, 2017-03-06:
@pan- .. any ideas? Thanks.

pan-, 2017-03-06:
@ashok-rao Apparently, google cut the support of URI beacon last august (see [here](https://github.com/google/physical-web/pull/774)). The update including this code hit the Google playstore last quarter. That's why URIBeacons are not visible anymore. 

> After 60 seconds or so, the beacon still is detected in config mode.
Using nRF connect, I see the mode of the beacon changing after 60 seconds.

I'm not sure of what we should do here. Maybe ending the life of this example is the right thing to do since the protocol has been abandoned and fully replace by Eddystone.

@sg- any thoughts ?

ashok-rao, 2017-03-06:
Oh, OK. Thanks @pan- . Looks like the URIBeacon is fully replaced by Eddystone. http://google.github.io/uribeacon/

sg-, 2017-03-06:
I'd suggest we remove that example.

MarceloSalazar, 2017-03-08:
@pan- could you please remove the URI beacon example in the OOB branch?

pan-, 2017-03-08:
Done, I close this.

PR #65 (2017-03-08): remove BLE_URIBeacon because the URI beacon protocol is outdated.
Fix #62
closed 2017-03-08 by pan-