ARMmbed/mbed-os-example-ble #75: Working with X-NUCLEO-IDB05A1?

mgiaco, 2017-04-27:
Hi, 
I am new to the BLE stuff. I would like to use the STM BLE module for some tests. I found this sample https://developer.mbed.org/teams/ST/code/BLE_HeartRate_IDB0XA1/ but the sample usese the old mbed os?
Is this repository the right starting point for the new mbed-os and BLE.  Is it possible that this examples also work with the X-NUCLEO-IDB05A1 shield?

many thanks
cheers
mathias

jmjobin, 2017-04-27:
The sample if I am not wrong gives you a bin. But you can do your updated version going to the On-line mbed compiler and IDE and import the "mbed-os-example-ble-HeartRate".
Here you should have the updated OS. This also use the 0XA1 shield
jean-marc

mgiaco, 2017-04-27:
Okay many thanks that was the right hint!
mathias

closed 2017-04-27 by mgiaco