ARMmbed/mbed-os-example-ble #76: BLE_UART service

mgiaco, 2017-04-27:
Hi, 
I would like to port a custom bootloader which uses UART actually.  So is there any UART Service Example around  like this one here: https://developer.mbed.org/teams/Bluetooth-Low-Energy/code/BLE_UARTConsole/

Sorry I am new to BLE and the new mbed-os, I am always wondering which one is for the old and which one is for the new mbed-os.

Or is there any better Service I can use?

jmjobin, 2017-04-27:
Not sure I understand all your questions and I will try some answers.
Probably someone may correct my answers.
There is no BLE UART service. You can do it but it must be custom. So make your own.
Read BLE specifications... :-)
For the mbed-os, do you mean which example use the old or new os?
I very often ask myself the same question :-D
You can check on the release by right clicking on the "mbed-os" directory and "revision".
Sometimes new versions may not work on the example, happens once in a while. But come on this forum and ask for corrections.
jean-marc

mgiaco, 2017-05-01:
Okay many thanks. 
Today I tried the example mentioned in this link here https://thejeshgn.com/2016/10/01/uart-over-bluetooth-low-energy/. Now I understand what you mean with your own protocol. 
So mbed-os uses this custom Uart Service. So give it a try: First I tried the BLE_HEARTRATE example to figure out everything works fine with my hardware. After that I replace the code of the main with the code from the example here https://developer.mbed.org/teams/Bluetooth-Low-Energy/wiki/UART-access-over-BLE.

Everything compiles with no error - nice :-) But when I try to connect with nRF UART v2.0 it doesn´t work. So I found the device but I always get the message "Device doesn't support UART.

Does someone has an idea what´s wrong? I do not think that the example depends on a NRF device or I am wrong here?

Is there any reason why there is no UART_SERVICE example here in this repository? 
many thanks.

jmjobin, 2017-05-01:
Hello, in my opinion, just an opinion, since the UART service is not a service but a custom you have to be sure you match the custom attribute from the nRF UART v2.0.
Maybe the Nordic forum will be able to answer to you. I never tried this yet. :-)
jean-marc

mgiaco, 2017-05-01:
Okay so I taught when the Uart Service is part of the code of mbed-os it should work.

jmjobin, 2017-05-01:
Well!!! This is a custom service :-) I use Spp with Bluetooth edr for uart. This is standard

mgiaco, 2017-05-02:
Okay but Spp is not available in BLE. I read in another post that you are using the NRF52_DK also. Please can you test if the UART Service is working with this HW?

many thanks

jmjobin, 2017-05-02:
Yes SPP is EDR (BT 2.0-3.0)
Sorry so far no time to test it :-D As soon as I can I will come back here but I think you will find a solution before. Let us know what you found.
jean-marc

mgiaco, 2017-05-02:
okay I will search for that. Is maybe someone else here who can help me a little bit?

mgiaco, 2017-05-05:
So it was a failure of mine. Now everything works as expected. Should I add a Pull Request for a UART_SERVICE example?

Here is the modified example based on the old example: https://developer.mbed.org/users/clonephone82/code/mbed-os-example-ble-UartService/
cheers
mathias

jmjobin, 2017-05-05:
Excellent, bravo... I do not know what you should do I am just a participant like you :-D

mgiaco, 2017-05-05:
:-) so the question was to the community here :-)

jmjobin, 2017-05-05:
Of course, sorry! Ahah

pan-, 2017-05-08:
@mgiaco Many thanks for sharing your work. It might be interesting to have an UART example. Please, feel free to make a PR.

closed 2018-06-01 by pan-