ARMmbed/mbed-os-example-ble #85: [OoB_5.5-RC2]: nRF52840_DK, ARMCC: Stack error.

ashok-rao, 2017-06-14:
Steps:
1. Clone repo, update to RC2 branch.
2. mbed compile -t ARM -m nRF52840_DK
3. Flash BLE_EddystoneService.hex to target.
4. Reset and run. Open a serial terminal connection to target.

Output on console:
```
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20003714, task name: )
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20003714, task name: )
```

SHA's:
mbed-os-example-ble (fc79579dfd21)
|- BLE_EddystoneService\mbed-os (162f80099d94)
|- BLE_EddystoneService\shields\TARGET_ST_BLUENRG (6670a4495aaf)

c1728p9, 2017-06-15:
Hi @ashok-rao can you test with https://github.com/ARMmbed/mbed-os/pull/4563 and see if it fixes your problem?

ashok-rao, 2017-06-22:
@c1728p9 .. Thanks. Yes, I can confirm that this fix works.. I'm closing this issue now.

@MarceloSalazar, FYI.

closed 2017-06-22 by ashok-rao