ARMmbed/mbed-os-example-ble #86: [OoB_5.5-RC2]: arm_beetle_soc, can not update the URL by app

toyowata, 2017-06-16:
Steps:
1. Clone repo, update to mbed-os-5.5.0-rc2-oob branch.
2. ```mbed compile -t ARM -m arm_beetle_soc```
3. Flash ```BLE_EddystoneService.bin``` to target.
4. Reset and run.
5. Start nRF Connect app
6. Scan device and connect to the EDDISTONE CONFIG device
7. Open Data part and write 0x0161726d00 value
8. Read the Value in Data, but it is empty (expected ```Value: https://www.arm.com```)
9. Disconnect from device, and scan again
10. Eddystone service is up, but the URL is www.mbed.com, not www.arm.com

SHA's:
```
BLE_EddystoneService (None)
|- mbed-os (162f80099d94)
`- shields\TARGET_ST_BLUENRG (6670a4495aaf)
```

pan-, 2017-06-16:
@toyowata I cannot reproduce the issue but I'm on it; the beetle board interface firmware doesn't work well on windows 10.

MarceloSalazar, 2017-06-28:
@pan- is there an updated on this?

pan-, 2017-06-28:
@MarceloSalazar Unfortunately, no, I've asked the team handling the beetle board but they have been unresponsive so far.

0xc0170, 2017-06-28:
cc @mmorenobarm

mmorenobarm, 2017-06-28:
cc @abhishek-pandit

MarceloSalazar, 2017-09-23:
@abhishek-pandit @mmorenobarm is there any update on this?

gabker01, 2017-10-30:
Issue was able to reproduced, fix is created and pull request is submitted: https://github.com/ARMmbed/mbed-os/pull/5400

closed 2017-11-03 by pan-