ARMmbed/mbed-os-example-ble #91: Export to Keil

noutram, 2017-06-27:
I just tried the HR example using a ST F401 board IBD05A1 peripheral board.
From the CLI, it seems to work fine which is nice.
When I export to uVision 5 and rebuild it however, (i) it does not work and (ii) I notice the LED is flashing faster.
Any ideas?

apalmieriGH, 2017-06-28:
Hi,
me and @pan-  are on the way and investigating on that...

noutram, 2017-06-28:
I should add that I am using the same ARM compiler for both CLI and Keil iVision (the version that comes with v5.23)

apalmieriGH, 2017-07-03:
Hi,
in the meantime you could try linking this commit for the BlueNRG library:

a6c43a45871b3c4eeafdac4a392f28e5faec98f4

noutram, 2017-07-03:
Broken link?

apalmieriGH, 2017-07-03:
Should work now

noutram, 2017-07-03:
Excellent - I'll give it a try when I get into the office this afternoon.

noutram, 2017-07-06:
Ok, I've now tried this and it seems to be working. Exported to uvision5, rebuild and deployed. I can now read HR measurements.

noutram, 2017-07-11:
@apalmieriGH Will there be a notification when this gets merged into the master branch?

apalmieriGH, 2017-07-11:
Yes, sure.

pan-, 2018-06-01:
Closing this as the issue as been resolved.

closed 2018-06-01 by pan-