ARMmbed/mbed-os-example-ble #92: Are the BLE examples provided here power efficient?

binary-nerd, 2017-07-04:
I wrote my code following the same lines as the examples here.

When tested on a new coin cell battery, in an hour or so, the battery is completely drained up and the BLE connection becomes unstable.

I would like to know if the code here is power optimized or is there scope for improvement?

The [following post](https://developer.mbed.org/forum/team-63-Bluetooth-Low-Energy-community/topic/5262/) describes that `ble.waitForEvent()` puts the ble into sleep in between events.. But we are not using it anywhere in the examples. Is it taken care of in other API's?

If there is a room for improvement, may be it is a good idea to write a fully power efficient BLE example which will act as reference to writing low power applications.

pan-, 2017-07-04:
@binary-nerd The examples are not to blame here, the high power consumption is caused by the OS. Until tickless mode is properly implemented, it will stay that way. 

However it is not impossible to tune your application to get the most of your battery. For example, we've worked closely with google to have a power efficient image of an Eddystone beacon ([here](https://github.com/roywant/EddystoneBeacon)): It can survives on a 1000mA battery for six years while advertising every 500ms.

To get to that result, the RTOS has been disabled and we makes sure that no peripheral stay open.

binary-nerd, 2017-07-05:
Thank you for the reference @pan- 

I will try to model my program following that application.

**BDW, Since the timeline for tickless mode support is not clear, is it recommended to use the ST provided BLE examples directly and then make the necessary changes to them instead of using the mbed framework?**

Also, I see things like configuring GapAdvertisingParameters and other stuff to manage power consumption. Isnt it good to have atleast 1 example in this repositpry showing how to make the BLE program power efficient? Also, I believe these things(approach followed in the Eddystone beacon example) would improve power consumption irrespective of tickless programming? No?

pan-, 2017-07-05:
> Also, I believe the approach followed in the Eddystone beacon example would improve power consumption irrespective of tickless mode?

It is the same approach as the one in the examples, both use event driven programming. The event loop used is not the same as the one bundled in mbed OS for historical reasons, not technical ones. 

> BDW, Since the timeline for tickless mode support is not clear, is it recommended to use the ST provided BLE examples directly and then make the necessary changes to them instead of using the mbed framework?

The is two approach you can follow if you want to optimize your application for power: 
* disable the RTOS if its not needed, a `.mbedignore` file similar to [this one](https://gist.github.com/pan-/e0e9ffb32ae734eda92a330b8f91d5f1) in your application root directory should be enough. 
* Add tickless mode yourself: you may find some informations [here](https://github.com/ARMmbed/mbed-os/issues/3509). Note that some API might have change with mbed OS 5.5.

jrobeson, 2018-03-05:
now that tickless is here, how does the advice change?

pan-, 2018-03-07:
@jrobeson I believe examples or configuration should not be changed and advices above should not be followed anymore.

closed 2018-06-11 by donatieng