ARMmbed/mbed-os-example-ble #121: nRF51 unresponsive for the BLE_Beacon example

gggekov, 2017-12-12:
## What is the problem
The nRF51 board is unresponsive for the `BLE_Beacon` example on the `mbed-os-5.7.0-oob` branch.
The board doesn't print anything on the serial port and the device doesn't appear on the Nordic mobile app. 

## What have I tried 
I monitor the serial port at 9600b/s and at 115200b/s - it doesn't help
I updated the firmware of the board - it doesn't help
I deleted and imported again the example repository - it doesn't help 

Do you have any suggestions what to do in order to make this example working?

Thanks and regards,
George

MarceloSalazar, 2017-12-12:
@pan- could you please have a look?
cc @adbridge

GarySwansonARM, 2017-12-13:
This works for me (BLE_LED and BLE_Battery) using the same board that I got from George, and GCC_ARM on the mac.

gggekov, 2017-12-13:
I'm still having problems with the BLE_Beacon example on nRF51 board on Windows.
The example worked straight away on nRF52 but on the nRF51 I don't get anything on the serial.

gggekov, 2017-12-13:
Thanks to @pan- and @GarySwansonARM  the problem is fixed.
The page of the nRF51 board(https://os.mbed.com/platforms/Nordic-nRF51-DK/) point to an old firmware that doesn't work on Windows10. Using the latest version of DAPLink solved the issue. 
The online doc needs to be updated to point to the latest version of DAPLink.

adbridge, 2017-12-14:
@AnotherButler Do you know how we get the online doc updated to point to the latest version of DAPLink ?

AnotherButler, 2017-12-14:
@mariannecrowder Do you update the content of the platforms page, or is there somewhere we can do this?

AnotherButler, 2017-12-14:
@adbridge I understand the PE team maintains these pages. @maclobdell Could you please help us with this?

MarceloSalazar, 2017-12-18:
@gggekov could you please confirm exactly which version is known to work ok?

gggekov, 2017-12-18:
I used the DAPLink Q3'17 - build v0244, the current latest release of DAPLink.

MarceloSalazar, 2017-12-18:
IOTPART-4233

closed 2018-06-11 by donatieng