ARMmbed/mbed-os-example-ble #128: Error: Cannot open source input file "ble/BLE.h": No such file or directory in "source/main.cpp", Line: 19, Col: 22

DuyTrandeLion, 2017-12-24:
I tried importing the new BLE example using the mbed online compiler but I received an error:
Error: Cannot open source input file "ble/BLE.h": No such file or directory in "source/main.cpp", Line: 19, Col: 22

Is this because of the compiler? Thank you

pan-, 2018-01-08:
@DuyTrandeLion Could you indicate which target and compiler yield the error ?

pan-, 2018-06-01:
Closing this; @DuyTrandeLion feel free to reopen if you continue to have troubles and/or wantsto provide more details.

closed 2018-06-01 by pan-

stevennoyce, 2019-08-19:
I'm having what I believe is the same issue.

It is fairly simple to replicate, as follows:

- In the mbed cloud IDE, import the `mbed-os-example-ble-HeartRate-PSoC` example
- Change the target to `PSoC 6 BLE Pioneer Kit`
- Press `Compile`
- Receive compilation error `'ble/BLE.h' file not found
- Switch target to `Nordic nRF52-DK`
- Press `Compile` again
- Build is now successful

I am new to the mbed ecosystem, but would be happy to help with this if given direction.

I have heard that a new `Future` version of the Cypress BLE HAL has been committed, but I do not know if that commit fixes this issue or when that version will become the default.

pan-, 2019-08-20:
IIRC BLE hasn't been enabled yet by Cypress on `PSoC 6 BLE Pioneer Kit`.

stevennoyce, 2019-08-24:
Ah, okay, that's unfortunate.  I didn't expect that, as I have a hard time seeing how a BLE kit can be advertised as mbed enabled if the mbed BLE interface is not functional on that board.

Is there somewhere I could check to see which boards are capable of running BLE via mbed?

pan-, 2019-08-27:
@steven-slatter [Here](https://github.com/ARMmbed/mbed-os/blob/7e179f5beb6946dd9221164d7082c38e82aa4fb3/targets/targets.json#L8632) you can see that the target doesn't have the BLE feature defined. It is not the [case](https://github.com/ARMmbed/mbed-os/blob/7e179f5beb6946dd9221164d7082c38e82aa4fb3/targets/targets.json#L8589) for some other targets from cypress.

pan-, 2019-08-28:
@stevennoyce I tagged the wrong user on my last comment but you can look it up for your answer.

stevennoyce, 2019-08-30:
I see.  Thanks for pointing me in the right direction.  I'm glad I now know where to look for that sort of thing.  Unfortunately digging through targets.json was not an obvious pathway for a new mbed user, but now I know.

Thanks again.