ARMmbed/mbed-os-example-ble #138: OOB: Bluetooth device fails to connect to running examples (NRF52_DK)

scartmell-arm, 2018-03-08:
Trying to run example applications `BLE_LED`, `BLE_LEDBlinker`, and `BLE_BatteryLevel` for 5.8 OOB testing using `mbed-os-5.8.0-oob` branch with suggested mbed-os branch `https://github.com/ARMmbed/mbed-os/#d1ba1089a31ffab6b6a8887092b3d1fcc7c23b72`

Connecting to the serial port of the device shows the correct output (BLE_LED in this case):

```
DEVICE MAC ADDRESS: e8:10:a0:6c:32:af
```

Connecting to the running device fails on NRF52_DK, however the same examples with the same configurations succeeds when running on an NRF51_DK.

[image]

scartmell-arm, 2018-03-08:
Using the default commit used in the example works correctly:
[https://github.com/ARMmbed/mbed-os/#6ff720302e43c7a420232d134e786811cc7281a9](https://github.com/ARMmbed/mbed-os/#6ff720302e43c7a420232d134e786811cc7281a9)

However, when changing to the commit suggested in the OOB form, the example fails to work:
[https://github.com/ARMmbed/mbed-os/#6ff720302e43c7a420232d134e786811cc7281a9](https://github.com/ARMmbed/mbed-os/#d1ba1089a31ffab6b6a8887092b3d1fcc7c23b72)

donatieng, 2018-03-08:
Thanks for reporting @scartmell-arm.

Seems like phones that support it ask for a PHY upgrade that we're not handling properly. Fix is on its way.

donatieng, 2018-03-08:
Should be fixed by this pull request: [Support for nRF52x BLE 4.2 and 5 events #6308](https://github.com/ARMmbed/mbed-os/pull/6308)

RogerCenson, 2018-03-31:
Hello! 
Was it fixed ? I am having the same problem here.

pan-, 2018-05-31:
I'm closing this as this as been fixed in https://github.com/ARMmbed/mbed-os/pull/6308

closed 2018-05-31 by pan-