ARMmbed/mbed-os-example-ble #157: NRF52 DK does not connect

mcprakash, 2018-06-08:
Hi 
 The NRF 52 DK (PCA 10040) s132 board does not work using ARM Mbed environment. The BLE_Battery_Level example sets the device to advertise as BATTERY_LEVEL but when I try to connect to it using my Samsung Galaxy 8+, it just disconnects and the device is not discover-able again.  The device works with KEIL examples with latest SDK 15/16. The ARM Mbed uses SDK 11 and firmware 2.0 whereas the latest firmware (SDK 15 v5 works fine). Would it be possible to update the SDK for the ARM Mbed environment for NRF52 DK ?

Thanks
Chandra

donatieng, 2018-06-08:
Hi @mcprakash - which version of mbed OS and which compiler are you using? This looks like #138 fixed by #6308.

mcprakash, 2018-06-08:
Hi @donatieng 
 I am using the latest mbed-os (from mbed init in the examples folder). I checked the commit to be #367dbdf5145f4d6aa3e483c147fe7bda1ce23a36 (mbed-os-5.8).
 Will check the commits on the link you have shared. 

Thanks
Chandra

pan-, 2018-06-12:
Hi @mcprakash 

Could you indicate which compiler you're using ? 

For your other request, in mbed os 5.9 (that will be released this month - it is on master) Nordic SDK14 will be used. Update may be made SDK16 at a latter date.

mcprakash, 2018-06-12:
Hi @pan- 
 I am using mbed-cli with GCC_ARM (latest version). I will try mbed-os-5.9 in the next day and get back to you. 

Best
Chandra

pan-, 2018-06-13:
@mcprakash @7211 may fix your issue; could you try it and let us know ?

mcprakash, 2018-06-13:
Hi @pan- 
 I tried mbed-os-5.9 and still getting the connection loss issue. It connects 1 time, don't see any updates on the phone and the device disappears (with reboot back to start). 
 Will try your fix and will report in a day.
Best
Chandra

mcprakash, 2018-06-14:
Hi @pan- 
 I tried you fix and still it does not connect to my phone. Will try with a different phone and see if it is a problem on the android side. 

Best
Chandra

closed 2018-12-19 by donatieng