ARMmbed/mbed-os-example-ble #164: Security Manager example questions/inconsistencies

yennster, 2018-07-02:
Coming from a developer perspective I have a few questions:

1. Why isn't the security manager example named: "BLE_SecurityManager" to be consistent with the folder naming convention of the other BLE examples?
1. In the context of the application, what is the "peripheral" and what is the "central"? Clarification in the [README](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_SM/README.md) for this would be helpful (i.e. for example, peripheral = phone, etc.)
1. Why are there blank brackets around the device run() functions in main() ([L404-L408](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_SM/source/main.cpp#L404-L408) & [L410-L414](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_SM/source/main.cpp#L410-L414))
1. Why is `mbed_events.h` included separately in addition to `mbed.h`? https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_SM/source/main.cpp#L17
1. From the README: "Upon success it will disconnect and start advertising to demonstrate usage as a peripheral." What will disconnect? Some clarification is required in the [README](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_SM/README.md)
1. The README could also use some images showing what a successful "connect" and "disconnect" look like, within the phone application and serial terminal

Thanks!

pan-, 2018-07-03:
> Why isn't the security manager example named: "BLE_SecurityManager" to be consistent with the folder naming convention of the other BLE examples?

I'm not sure to completely understand how naming is inconsistent. We do use abbreviations to name Bluetooth protocols: BLE_GAP is not named BLE_GenericAccessProtocol, BLE_GattClient is not named BLE_GenericAttributeProtocolClient. 

> In the context of the application, what is the "peripheral" and what is the "central"? Clarification in the README for this would be helpful (i.e. for example, peripheral = phone, etc.)

I suppose we can clarify what roles are.

> Why are there blank brackets around the device run() functions in main()? (L404-L408 & L410-L414)

That's C/C++ syntax to define a scope. So that the `SMDevice*` defined in the scope gets destroyed after execution. 

> Why is mbed_events.h included separately in addition to mbed.h? (L17)

I think it is a copy and paste mistake from a time where events wasn't part of mbed-os . We can fix this.

> From the README: "Upon success it will disconnect and start advertising to demonstrate usage as a peripheral." What will disconnect? Some clarification is required in the README
The README could also use some screenshots showing what a successful "connect" and "disconnect" look like, within the phone application and serial terminal

I agree we can clarify the README. Thanks again for the feedback. However I'm not so sure about screenshot of the phone application as those can gets deprecated real quick.

yennster, 2018-07-03:
@pan- Cool, thanks for answering my questions!

> I agree we can clarify the README. Thanks again for the feedback. However I'm not so sure about screenshot of the phone application as those can gets deprecated real quick.

I still feel like the README should have some sort of indication what a successful connection and disconnection look like, maybe just a copy/paste from the terminal output?

closed 2019-09-19 by yennster