ARMmbed/mbed-os-example-ble #169: Does not compile with nRF52-DK

Navis-Raven, 2018-08-10:
When selecting BLE_Button for nRF52-DK, the programm does not compile, and the hardware included is not the good one (ours is nRF52832 and not nRF51822)

[image]

pan-, 2018-08-13:
Are you using https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-Button/ ? 
That is the version we maintain and that is synced with this repository.

Navis-Raven, 2018-08-13:
Select the nRF52-DK as a hardware one Mbed compiler, then new, at just after the BLE_BUTTON is selected automatically, and this is the software which has a problem. Maybe this is not the version you maintain, I don't know

pan-, 2018-08-13:
I've just tried the version I linked; it compiles without trouble. Please try it; that is the only version we maintain.

Navis-Raven, 2018-08-13:
So this is not the problematic version.

Navis-Raven, 2018-08-13:
If the other version is not maintained, so you have to remove it from the default example list and replace it by this one.

ConradBraam, 2019-02-14:
[image]
Hi Navis-Raven , I managed to reproduce. I see that the "new project" wizard is importing a project with the same name which is MBED Version 2. You do not see the same projects available here, because the wizard is picking up ones the vendor/target originally added at the time.
We want to use mbed 5.X  for new designs, else you will not be getting support for any new features and getting much assistance I imagine. The browser list probably should show you V5 projects in that list, but mbed don't control that list, so the way to go is to **import the program from the mbed examples, then choose your target**. This gets you the new example.

I hope these steps help the next person.
1. go to the new project https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-Button/ , and tap the "import into compiler" button
2. Choose a unique program name, then change your target to the one you really want. I choose the STD52832B as a challenge. 
3. The compile fails, because this target does not have a button built in, so you will have to wire one in, I chose to use GPIO0 (which is an LED actually but just using it so it will compile)
 So I edit the mbed_app.json file, and add these lines to support the target
"SDT52832B": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S132_FULL", "NORDIC_SOFTDEVICE"],
            "ble_button_pin_name": "GPIO0"
        },
4. Hit compile again.
If these steps do work for you, let me know, so I can "attempt" to fix the flow for future users.

Note: I have not verified that the SDT52832B actually works, just laying the steps logically I hope.

Navis-Raven, 2019-02-15:
@ConradBraam hi, thanks for your answer, but I cannot make a test anymore since à I returned back the nRF52-DK board to it's owner since September 2018. 

However, what I point here is that it has to work from the beginning without anymore else configuration. So the Arm Mbed Compiler has te edit json file and all configuration files in order to generate the Keil project without any additionnal configuration.

ConradBraam, 2019-02-15:
No problem about the NRF52. Your main issue was compile-fail, for compiling code no longer in support. So I have asked web team to fill the examples list with all the new mbed5 examples instead :-) (Should happen early next week.) That should prevent the original issue - but not your JSON and Keil export issue.

I think we need to remember that the JSON editing is a larger problem because it requires keeping the "app level" configuration data for over a dozen examples all in sync, so that's why we limit the amount of targets we explicitly support, it's often easy enough to copy-paste once you know what you are doing. Which is why I explained the json file editing in a bit of detail above. We have over 100 targets supported. Targets are themselves just templates for dev boards, which get ignored when moving to production anyway, so of less value to anything in a real application. If there are ways we can simplify the JSON editing process with better documentation, do let me know, docs are easy to change.

My only advice on getting projects exported into Keil to run offline is to start over from scratch and have a lot of Keil experience already. I have had projects export but require minor perfectly simple settings changes on older builds. But the examples are just that, a starting place to learn from, a known good reference point, and that you need to be prepared to throw away and fully design an app instead. The examples are often not scaleable. So fresh start is best, but yes, the examples must actually work 100%!

I kind of get that you are raising this as a way of getting mbed to make things easier, I hope that removing the old examples and putting the many more new ones into the examples list for NRF52 will now solve some of that. Users also want to look out for a new mbed Studio coming soon. https://os.mbed.com/docs/mbed-os/v5.11/tools/developing-mbed-studio.html It will support online and offline (win/gnu-linux/mac) using arm compiler I believe.

Navis-Raven, 2019-02-15:
@ConradBraam  if facts I've since then, I've uninstalled Keil and abandonned this editor, and I switched to VSCode with PlatformIO plugin, it install and configure all you want with code example for more than... 500 boards (not only ARM based).

The good thing is tha PlatformIO is open source, I don't know but maybe you can find your answer of how generating some config file for these boards... 

"My only advice on getting projects exported into Keil to run offline is to start over from scratch and have a lot of Keil experience already."
That's the whole problem... The main problem of beginning a new project is configuration. And if you don't have "this lot of Keil experience already" you're dead. Note that I had a similar problem with IAR and arm online compiler which was not up to date. 

The online arm compiler should have to simplify the configuration by offering this possibility of exporting with "zero configuration" (or near zero configuration)

"want to look out for a new mbed Studio coming soon." At last, a true arm mbed studio.... I'see that is based on eclipse. Unfortunately the arm tools have many bugs (not only this one), and I had many problems with them, and the developpement and bug correction is very slow . As I said before, I prefered tos switch to VSCode with platformIO there is less bugs and zero configuration (except the driver installation). 
https://platformio.org
https://code.visualstudio.com

ConradBraam, 2019-02-18:
Just want to correct something, the new mbed studio in alpha currently is not Eclipse based, it's java, but one of the new frameworks which allow running in browser and offline all in one familiar app. It should also be making OpenOCD debug setup painless. About 10 years ago I worked with a bloke who only coded in a plain text editor, he had systemized and memorized all the names he needed to and intelli-sense was not necessary, it was in his head. He only used the IDE for debugging. It's a very context sensitive art, and my fave right now it CLion, I'll be buying a license soon. 

Once I have solved the problem with old examples being presented to users, I will close this ticket off. There are a whole load of new examples that will appear in that list shortly. Thanks for highlighting this. Since that was the original issue, one issue per ticket.

Navis-Raven, 2019-02-18:
"is not Eclipse based, it's java"
Eclipse is itself based on Java.

But  it seems that you're right, since there is some updates
[image]

Navis-Raven, 2019-02-18:
I don't know about CLion but now I've found a very good IDE with very good plugin which facilitates all thinggs, and it is PlatformIO

Navis-Raven, 2019-02-18:
There is an arm mbed extension for vscode but I think not as good as PlatformIO

[image]

Navis-Raven, 2019-10-31:
@ConradBraam what's new about this bug ?

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

pan-, 2020-10-02:
I'm closing this. It is unrelated to the examples in this repository.

closed 2020-10-02 by pan-

Navis-Raven, 2020-10-02:
> I'm closing this. It is unrelated to the examples in this repository.

That's false. Off course it was related to the examples in the repository (see below)
[image]

I don't know how your project evolved since 2018, if after two years I declared this bug, it's the only answer you give... it is quite revolting for the users.