ARMmbed/mbed-os-example-ble #175: [OOB 5.10] Issue building BLE_BatteryLevel with online compiler

bridadan, 2018-09-05:
### Steps taken

Using production compiler and partner beta, the following issue is reproducible:

1. Import **this repository** (not the mercurial mirror on os.mbed.com) into online compiler
2. Checkout to branch `mbed-os-5.10.0-oob`
3. Delete all directories except `BLE_BatteryLevel`
4. Select target as NRF52_DK
5. Click compile button

I received the following error: 

> Error: Cannot open source input file "nrf_error.h": No such file or directory in "extras/BLE_BatteryLevel/mbed-os.lib/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/libraries/util/sdk_errors.h", Line: 73, Col: 24

It may be helpful to note that the mirror repository without the 5.10.0 updates works fine: https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-BatteryLevel

pan-, 2018-09-06:
I've tried to compile the project locally and it seems to work just fine. 
Do you know how the shenanigans to compile in the online compiler impact sources ? 

One solution would be to have a special OOB organisation on developer.mbed.org that host projects on OOB branches. I suppose the synchronization script can be useful in that task.

bridadan, 2018-09-06:
@pan- It works locally, its specifically an issue with something in the online compiler.

@theotherjimmy based on the error above do you have any tips to try/are you aware of any existing issues?

MarceloSalazar, 2018-09-13:
Opening to keep track of issues

pan-, 2018-09-13:
@MarceloSalazar  To keep track of issue, that would be better to open it in the right place. 
That's not useful here and it won't be fixed by us.

soleilplanet, 2018-09-19:
original error not seen, but new error message observed:
Error: #error directive: [NOT_SUPPORTED] No network configuration found for this target. in "extras/BLE_SM/mbed-os.lib/TEST_APPS/device/socket_app/cmd_ifconfig.cpp", Line: 27, Col: 3
Both BLE_SM and BLE_BatteryLevel failed with the same error message.

naveenkaje, 2018-11-13:
I have seen similar error message when I have a typo in my mbed_app.json file

```
{
    "target_overrides": {
        "NRF52_DK": {
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S132_FULL"],
            "target.extra_labels_add": ["SOFTDEVICE_MBR"] 
        }
    }
}
```
instead of
```
{
    "target_overrides": {
        "NRF52_DK": {
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S132_FULL"],
            "target.extra_labels_add": ["SOFTDEVICE_S132_MBR"] 
        }
    }
}
```
produces

```
Compile [  0.6%]: mbed_tz_context.c
Compile [  0.7%]: at24mac.cpp
[Error] sdk_errors.h@73,0:  #5: cannot open source input file "nrf_error.h": No such file or directory
[ERROR] "./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/libraries/util/sdk_errors.h", line 73: Error:  #5: cannot open source input file "nrf_error.h": No such file or directory
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\at24mac.cpp: 0 warnings, 1 error

[mbed] Auto-installing missing Python modules...
[mbed] ERROR: "C:\Python27\python.exe" returned error code 1.
[mbed] ERROR: Command "C:\Python27\python.exe -u C:\mbed-cli\mbed-hacks\NRF_serial_test\mbed-os\tools\make.py -t ARM -m NRF52_DK --source . --build .\BUILD\NRF52_DK\ARM" in "C:\mbed-cli\mbed-hacks\NRF_serial_test"
```
Just another place to check.

donatieng, 2018-12-07:
Closing as it's related to the online compiler - please raise an issue here if you still see it: https://github.com/ARMmbed/mbed-os/issues

closed 2018-12-07 by donatieng