ARMmbed/mbed-os-example-ble #177: [OOB 5.10] API documentation in os.mbed.com does not render correctly in Microsoft Edge

soleilplanet, 2018-09-06:
### Steps Taken
1. visit https://os-doc-builder.test.mbed.com/docs/development/apis/bluetooth.html with Microsoft Edge
2. all pages reference to wrong pages
    e.g. BLE class reference -> pointed to index.html
    e.g. GAP -> pointed to network-socket.html
    e.g. iBeacon -> points to example ble heart rate monitor's page

However, everything runs well in Chrome.

cmonr, 2018-09-06:
@soleilplanet Do you mind attaching images? I tried to reproduce locally and didn't find any issues across Firefox, Chrome, and Edge. (Browser versions and OS version would help in reproducing the issue)

pan-, 2018-09-11:
I'm closing this as references looks fine (if they are not it should be reported to https://github.com/ARMmbed/mbed-os-5-docs). 
Rendering on edge is fine too; if not, please report to https://github.com/ARMmbed/os.mbed.com .

closed 2018-09-11 by pan-