ARMmbed/mbed-os-example-ble #179: Thermometer link broken

JanneKiiskila, 2018-09-11:
Link to: 

https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.health_thermometer.xml

does not work - reference is in: https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_Thermometer/readme.md#thermometer

JanneKiiskila, 2018-09-11:
Working link (now at least) - https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.health_thermometer.xml&u=org.bluetooth.service.health_thermometer.xml

JanneKiiskila, 2018-09-11:
https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicViewer.aspx?u=org.bluetooth.characteristic.temperature_type.xml is broken, too.

JanneKiiskila, 2018-09-11:
Actually, all bluetooth.org links are down...

pan-, 2018-09-11:
The only one that doesn't works for me is https://developer.bluetooth.org/gatt/profiles/Pages/ProfileViewer.aspx?u=org.bluetooth.profile.health_thermometer.xml . Others looks fine.

PR #189 (2018-11-07): Fix link to health thermometer GATT service
Fixes #179
closed 2018-11-07 by pan-