ARMmbed/mbed-os-example-ble #182: How to use K64 board with ST shield(X-NUCLEO-IDB05A1)

XiangcaiHuang, 2018-09-26:
Following the main README:
```
$ cd mbed-os-example-ble
$ cd BLE_Beacon
$ mbed deploy
$ mbed compile -t GCC_ARM -m K64F
```

Download the bin file to K64F, but encounter a runtime error when it run to `BLE &ble = BLE::Instance();`:
**++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x13373
Error Value: 0x0
Current Thread: Id: 0x2000290C Entry: 0x1BBB5 StackSize: 0x1000 StackMem: 0x20002958 SP: 0x20003868

-- MbedOS Error Info --
Please provide an implementation for the HCI driver**

I have found the file, BlueNrgHCIDriver.cpp had not been compiled. Can you tell me how to solve it?

pan-, 2018-09-26:
@XiangcaiHuang It is detailed in the [readme](https://github.com/ARMmbed/mbed-os-example-ble#using-st-nucleo-shield-on-other-targets). If an [hardware modification](https://os.mbed.com/teams/ST/code/X_NUCLEO_IDB0XA1/) is necessary; apply it.

0xc0170, 2018-09-26:
@XiangcaiHuang Does readme has all to answer your question?

XiangcaiHuang, 2018-09-29:
@pan- 
Hi, Can you tell me how to apply the hardware modification?

XiangcaiHuang, 2018-09-29:
@0xc0170 
I don't know how to apply the hardware modification, there is no detailed description.

donatieng, 2018-10-01:
@XiangcaiHuang Links shared by @pan- should help you with both of these.

XiangcaiHuang, 2018-10-11:
Hi, I have made a hardware switch by moving zero resistor R4 to R6 so that D13 is now the SPI clock. But it show me the same runtime error.

donatieng, 2018-12-17:
@XiangcaiHuang you need to apply the configuration suggested in the README file - this would look like:
```json
"K64F": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_BLUENRG"]
        },
```

Please re-open if that does not resolve this.

closed 2018-12-17 by donatieng