ARMmbed/mbed-os-example-ble #198: No "Connect" option: BLE HearRate Example

naveenkaje, 2018-12-05:
I am following the steps described in the README.md on NRF52840_DK. The Heartrate monitor is seen during scanning. However, it does not give a "Connect" Option.
See screenshot attached.

I also tried altering the adv_pameters ble::adevertising_type_t::CONNECTABLE_UNDIRECTED to adv_pameters ble::adevertising_type_t::CONNECTABLE_DIRECTED. That did not help.

[image]

donatieng, 2018-12-06:
Should be fixed in RC2 by https://github.com/ARMmbed/mbed-os/pull/8884, will confirm

pan-, 2018-12-06:
The fix is actually here: https://github.com/pan-/mbed/commit/0543442cc315b1d25c674b309230753fd99e1f6a . 
We gathering other fixes in that branch and will submit it to Mbed-os shortly.

closed 2018-12-10 by cmonr