ARMmbed/mbed-os-example-ble #199: NRF52_DK: HeartRate Example: MbedOS Assertion Failed

naveenkaje, 2018-12-05:
I used mbed-os-5.11.0-oob branch of mbed-os-example-ble-HeartRate with mbed-os (mbed-os-5.11.0-rc1)
Error printed from NRFCordioHCIDriver.cpp
``` 
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: _stack_buffer != NULL
Location: 0x19393
File: .\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC\TARGET_NORDIC_CORDIO\TARGET_NRF5x\NRFCordioHCIDriver.cpp+154
Error Value: 0x0
Current Thread: main  Id: 0x20007D7C Entry: 0x1A135 StackSize: 0x1000 StackMem: 0x20007DC0 SP: 0x20008BCC
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0144
 -- MbedOS Error Info --
```

Built with the following command using GCC toolchain

```
mbed compile -m NRF52_DK -t GCC_ARM
```

donatieng, 2018-12-06:
Thanks @naveenkaje - fix is here:
https://github.com/ARMmbed/mbed-os/pull/8981

adbridge, 2018-12-13:
@donatieng can we close this then ?

donatieng, 2018-12-17:
Resolved, closing

closed 2018-12-17 by donatieng