ARMmbed/mbed-os-example-ble #201: NRF52840_DK: HeartRate Example: Compiling failed with ARM toolchain

soramame21, 2018-12-12:
I got a compilation error for NRF52840_DK when running OOB test with following steps.

```
C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate>git log
commit dfa6dbad946af889291de25f6530a07a80087eb7 (HEAD -> mbed-os-5.11.0-oob, origin/mbed-os-5.11.0-oob, origin/HEAD)
Author: Marcelo Salazar <marcelo.salazar@arm.com>
Date:   Thu Dec 6 14:44:07 2018 +0000

    Update BLE HeartRate app to 5.11-RC2

commit a6572d0d050f5dcfec21ff4b99ca729d10281ee8
Author: Marcelo Salazar <marcelo.salazar@arm.com>
Date:   Mon Dec 3 12:34:03 2018 +0000

    Update BLE Heartrate app to 5.11-RC1

C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate>mbed compile -m NRF52840_DK -t ARM -vv -f
;;;
[DEBUG] Output: .\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5x\us_ticker.c: 3 warnings, 0 errors
Compile [100.0%]: main.cpp
Compile: C:\Keil_v5\ARM\ARMCC\bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M4.fp --md --no_depend_system_headers --c99 -D__ASSERT_MSG --cpp --no_rtti --no_vla -DMBED_ROM_START=0x0 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x40000 -DFEATURE_CRYPTOCELL310=1 -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DDEVICE_SYSTICK_CLK_OFF_DURING_SLEEP=1 -D__FPU_PRESENT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_MPU=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_ITM=1 -DTARGET_SOFTDEVICE_NONE -DMBEDTLS_CONFIG_HW_SUPPORT -DNRF52840_XXAA -DDEVICE_USTICKER=1 -DWSF_MAX_HANDLERS=10 -DTARGET_NRF52840_DK -DTARGET_NORDIC_CORDIO -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_TRNG=1 -DTARGET_M4 -DDEVICE_QSPI=1 -DFEATURE_BLE=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DTARGET_NRF5x -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DTARGET_LIKE_MBED -DCOMPONENT_PSA_SRV_EMUL=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_SDK_14_2 -DTARGET_CORDIO -DTARGET_NORDIC -DTARGET_CORDIO_LL -DDEVICE_SERIAL_FC=1 -DCONFIG_GPIO_AS_PINRESET -DBOARD_PCA10056 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DMBED_BUILD_TIMESTAMP=1544583552.15 -DNRF52_ERRATA_20 -DDEVICE_SPI=1 -DCOMPONENT_NSPE=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DTOOLCHAIN_ARM_STD -DTARGET_MCU_NRF52840 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_NRF52840 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DARM_MATH_CM4 -DTARGET_NRF52 --preinclude=.\BUILD\NRF52840_DK\ARM\mbed_config.h --via .\BUILD\NRF52840_DK\ARM\.includes_79773019d3251acc58ba13de3c0d29bb.txt --depend BUILD\NRF52840_DK\ARM\source\main.d -o BUILD\NRF52840_DK\ARM\source\main.o .\source\main.cpp
[Error] main.cpp@131,0:  #135: namespace "ble"  has no member "DisconnectionEvent"
[Warning] main.cpp@131,0:  #997-D: function "ble::Gap::EventHandler::onDisconnectionComplete(const ble::DisconnectionCompleteEvent &)"  is hidden by "HeartrateDemo::onDisconnectionComplete"  -- virtual function override intended?
[Warning] Duration.h@237,0:  #186-D: pointless comparison of unsigned integer with zero
[Warning] Duration.h@237,0:  #186-D: pointless comparison of unsigned integer with zero
[DEBUG] Return: 1
[DEBUG] Output: ".\source\main.cpp", line 131: Error:  #135: namespace "ble"  has no member "DisconnectionEvent"
[DEBUG] Output: ".\source\main.cpp", line 131: Warning:  #997-D: function "ble::Gap::EventHandler::onDisconnectionComplete(const ble::DisconnectionCompleteEvent &)"  is hidden by "HeartrateDemo::onDisconnectionComplete"  -- virtual function override intended?
[DEBUG] Output: "./mbed-os/features/FEATURE_BLE/ble/common/Duration.h", line 237: Warning:  #186-D: pointless comparison of unsigned integer with zero
[DEBUG] Output:           detected during:
[DEBUG] Output:             instantiation of "Rep ble::Duration<Rep, TB, Range, Forever>::clamp(Rep) [with Rep=std::uint32_t, TB=1000U, Range=ble::Range<0U, 4294967295U>, Forever=void *]"  at line 120
[DEBUG] Output:             instantiation of "ble::Duration<Rep, TB, Range, Forever>::Duration(Rep) [with Rep=std::uint32_t, TB=1000U, Range=ble::Range<0U, 4294967295U>, Forever=void *]"  at line 69 of ".\source\main.cpp"
[DEBUG] Output: "./mbed-os/features/FEATURE_BLE/ble/common/Duration.h", line 237: Warning:  #186-D: pointless comparison of unsigned integer with zero
[DEBUG] Output:           detected during:
[DEBUG] Output:             instantiation of "Rep ble::Duration<Rep, TB, Range, Forever>::clamp(Rep) [with Rep=std::uint16_t, TB=10000U, Range=ble::Range<0U, 65535U>, Forever=ble::Value<std::uint16_t, (std::uint16_t)0U>]"  at line 120
[DEBUG] Output:             instantiation of "ble::Duration<Rep, TB, Range, Forever>::Duration(Rep) [with Rep=std::uint16_t, TB=10000U, Range=ble::Range<0U, 65535U>, Forever=ble::Value<std::uint16_t, (std::uint16_t)0U>]"  at line 231
[DEBUG] Output:             instantiation of "ble::Duration<Rep, TB, Range, Forever> ble::Duration<Rep, TB, Range, Forever>::forever() [with Rep=std::uint16_t, TB=10000U, Range=ble::Range<0U, 65535U>, Forever=ble::Value<std::uint16_t, (std::uint16_t)0U>]"  at line 624 of "./mbed-os/features/FEATURE_BLE/ble/gap/Gap.h"
[DEBUG] Output: .\source\main.cpp: 3 warnings, 1 error
Traceback (most recent call last):
  File "C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate\mbed-os\tools\make.py", line 73, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate\mbed-os\tools\build_api.py", line 540, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate\mbed-os\tools\toolchains\__init__.py", line 413, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate\mbed-os\tools\toolchains\__init__.py", line 476, in compile_queue
    raise ToolException(err)
ToolException: ".\source\main.cpp", line 131: Error:  #135: namespace "ble"  has no member "DisconnectionEvent"
".\source\main.cpp", line 131: Warning:  #997-D: function "ble::Gap::EventHandler::onDisconnectionComplete(const ble::DisconnectionCompleteEvent &)"  is hidden by "HeartrateDemo::onDisconnectionComplete"  -- virtual function override intended?
"./mbed-os/features/FEATURE_BLE/ble/common/Duration.h", line 237: Warning:  #186-D: pointless comparison of unsigned integer with zero
          detected during:
            instantiation of "Rep ble::Duration<Rep, TB, Range, Forever>::clamp(Rep) [with Rep=std::uint32_t, TB=1000U, Range=ble::Range<0U, 4294967295U>, Forever=void *]"  at line 120
            instantiation of "ble::Duration<Rep, TB, Range, Forever>::Duration(Rep) [with Rep=std::uint32_t, TB=1000U, Range=ble::Range<0U, 4294967295U>, Forever=void *]"  at line 69 of ".\source\main.cpp"
"./mbed-os/features/FEATURE_BLE/ble/common/Duration.h", line 237: Warning:  #186-D: pointless comparison of unsigned integer with zero
          detected during:
            instantiation of "Rep ble::Duration<Rep, TB, Range, Forever>::clamp(Rep) [with Rep=std::uint16_t, TB=10000U, Range=ble::Range<0U, 65535U>, Forever=ble::Value<std::uint16_t, (std::uint16_t)0U>]"  at line 120
            instantiation of "ble::Duration<Rep, TB, Range, Forever>::Duration(Rep) [with Rep=std::uint16_t, TB=10000U, Range=ble::Range<0U, 65535U>, Forever=ble::Value<std::uint16_t, (std::uint16_t)0U>]"  at line 231
            instantiation of "ble::Duration<Rep, TB, Range, Forever> ble::Duration<Rep, TB, Range, Forever>::forever() [with Rep=std::uint16_t, TB=10000U, Range=ble::Range<0U, 65535U>, Forever=ble::Value<std::uint16_t, (std::uint16_t)0U>]"  at line 624 of "./mbed-os/features/FEATURE_BLE/ble/gap/Gap.h"
.\source\main.cpp: 3 warnings, 1 error

[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate"
       Command: "C:\Python27\python.exe -u C:\ioT\511ooB\rc2\mbed-os-example-ble-HeartRate\mbed-os\tools\make.py -t ARM -m NRF52840_DK --source . --build .\BUILD\NRF52840_DK\ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

cmonr, 2018-12-12:
@ARMmbed/mbed-os-pan Fyi.

@soramame21 When RC3 comes out, (a couple of hours from now), could you recheck to make sure this is an issue?

soramame21, 2018-12-12:
@cmonr Thanks for such quick comment. Yes, I'd like to recheck with RC3.

cmonr, 2018-12-12:
Keep an eye on this PR: https://github.com/ARMmbed/mbed-os/pull/9064

Once it's in, PRs to example repos (including this one) will be opened to update the mbed-os.lib. Once that's merged, you can retest.You could also test right now by entering into the mbed-os repo in the

pan-, 2018-12-12:
For a start do not use Marcelo's repo. It was expected it wouldn't be synced and it is not ... 
Between RC1 and RC2 we applied API changes in mbed OS that were back ported to the example repo after the release of RC2.

soramame21, 2018-12-14:
@pan- @cmonr Which repo should I test? the marcelo's repo has been updated?

cmonr, 2018-12-14:
@soramame21 I would not suggest using @MarceloSalazar's fork, as all of the fixes should be landing back in this repo.

donatieng, 2018-12-17:
Closing as it was fixed and validated on our end, feel free to re-open if not

closed 2018-12-17 by donatieng