ARMmbed/mbed-os-example-ble #211: BLE_GattServer HardFault using BlueNRG and F401RE

KyleHunter, 2019-01-15:
Upon connecting to the server with a phone, MbedOS crashes when using the BLE_GattServer example:

Ble process started.
Ble instance initialized
Advertising started.
Adding demo service
clock service registered
service handle: 11
.hour characteristic value handle 13
.minute characteristic value handle 16
.second characteristic value handle 19
Connected.

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000000
R1   : 200035F8
R2   : 20018000
R3   : 00000000
R4   : 00000001
R5   : 00000000
R6   : 20000E6C
R7   : 00000000
R8   : 0000000A
R9   : 20000E6C
R10  : FFFFFFFF
R11  : 0000007B
R12  : 200035F8
SP   : 200035F8
LR   : 080203EF
PC   : 0800D4EA
xPSR : 81000000
PSP  : 20003590
MSP  : 20017FD8
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000008
AFSR : 00000000
BFAR : 20018024
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

pan-, 2019-01-15:
Thanks for your report @KyleHunter . Could you provide the following informations: 
* What version of mbed OS do you use ? 
* What version of examples do you use ? 
* What compiler do you use ? 
* What compilation profile do you use ? 
* Which target do you use ?

KyleHunter, 2019-01-15:
What version of mbed OS do you use ? Revision 5345:c966348
What version of examples do you use ? Latest from github
What compiler do you use ? Online mbed
What compilation profile do you use ? Huh? Just compile from online
Which target do you use ? F401RE with a IDB05A1 BlueNRG daughterboard

KyleHunter, 2019-01-16:
@ciarmcom Huh? That link is broken

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

pan-, 2020-05-29:
@KyleHunter I tried to reproduce the issue locally but everything seems to work with the latest version of mbed-os.

pan-, 2020-10-02:
@KyleHunter I'm closing this since we didn't received update from you. Please reopen if it is still an issue.

closed 2020-10-02 by pan-