ARMmbed/mbed-os-example-ble #224: why some example don't have destruct function?

bestony, 2019-03-21:
The Example `BLE_BatteyLevel` didn't have the  destruct function.

https://github.com/ARMmbed/mbed-os-example-ble/blob/5f4cead86f34900a99096070b0f5a07ea8a9e11e/BLE_BatteryLevel/source/main.cpp#L31-L50

same as `BLE_Beacon`,`BLE_Button`.

The BLE_GAP has destruct function.

https://github.com/ARMmbed/mbed-os-example-ble/blob/5f4cead86f34900a99096070b0f5a07ea8a9e11e/BLE_GAP/source/main.cpp#L99-L120

pan-, 2019-03-21:
@bestony I believe it is an harmless mistake. If you look at both examples, destructor is never called as we never exit the main. @paul-szczepanek-arm Any opinion about this issue ?

bestony, 2019-03-21:
yes, it harmless. but i think this is a exmaple for people who learn Mbed OS.

The cpp is diffcult to control Pointer , Keep Strict maybe is better for people who write code in embedded device

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@bestony It appears your question has been answered. Please reopen if you have further questions.

bestony, 2019-05-01:
@linlingao Close is OK , but i think no destruct function will make more and more developer write some bad smell code.