ARMmbed/mbed-os-example-ble #237: U-BLOX Nina B112

tcpipchip, 2019-05-15:
I compiled (mbed-cli) the LED APPLICATION example to U-BLOX NINA B112 and execute it, but the BLE SCANNER didnt find it. But sends data to serial and blink led

mbed compile -t GCC_ARM -m UBLOX_EVK_NINA_B1 

then i burned the BLE_LED.hex

I remember that this example had worked in the past!

[image]

Same with the mbed-compiler online version....

[image]

Fails too!

I tested on u-blox evb and module stand alone

Can you verify ?

Thanks

tcpipchip, 2019-05-15:
One tip, 

I did find some hex files created with mbed-cli weeks ago (wich includes the BLE LED example) and they works very fine after i burn on NINA B112.

Is it a MBED-OS update problem to NINA B112 ?

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

pan-, 2020-10-01:
@tcpipchip Is this issue still relevant ? The U-BLOX Nina B112 isn't supported by Mbed OS 6.

tcpipchip, 2020-10-01:
ok! Thank you!

closed 2020-11-10 by paul-szczepanek-arm