ARMmbed/mbed-os-example-ble #248: Incorrect NULL pointer check - Gatt Client Example

deepikabhavnani, 2019-06-27:
https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_GattClient/source/main.cpp#L466

```
                                    ^
mbed-os\platform/mbed_toolchain.h(352): note: expanded from macro 'MBED_DEPRECATED'
#define MBED_DEPRECATED(M) __attribute__((deprecated(M)))
                                          ^
./source/main.cpp(466): error: comparison between pointer and integer ('GattClientProcess::DiscoveredCharacteristicNode *' and 'bool')
        if (new_node == false) {
            ~~~~~~~~ ^  ~~~~~
9 warnings and 1 error generated.
compiling main.cpp...
```

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

pan-, 2020-05-29:
Fixed: https://github.com/ARMmbed/mbed-os-example-ble/blob/eaccb488a28794b5dbc8034020af91320f787f67/BLE_GattClient/source/main.cpp#L484

closed 2020-05-29 by pan-