ARMmbed/mbed-os-example-ble #257: Security Manager example broken for nRF52840_DK

AGlass0fMilk, 2019-08-17:
Hi,

While attempting to build and test the `BLE_SM` example on the `nRF52840_DK` target, my phone is able to connect but the Mbed-OS device encounters an error when attempting to upgrade the link's security. The following debug output is shown:

```
 PERIPHERAL                                                                     
                                                                                
[XXXXXXXX]                                                              
Please connect to device                                                        
Connected to peer [XXXXXXX]                                        
Error during SM::setLinkSecurity 4                                              
Advertising timed out - aborting 
```

The `SM:setLinkSecurity` call is returning `BLE_ERROR_INVALID_PARAM` at this line:

https://github.com/ARMmbed/mbed-os/blob/033fffea8406fc32b4a19d92d76912c86b010f86/features/FEATURE_BLE/source/generic/GenericSecurityManager.tpp#L367-L370

`get_control_blocks` loops through a number of connection control blocks trying to find the one given by the caller. None of the control blocks are in the `connected` state yet -- so the call fails to find the control block and thus the security manager returns `Invalid Parameter`.

My theory is that the internal state hasn't been updated before the stack's `onConnectionComplete` handler is called.

I'm going to try and delay the link security upgrade by deferring to the example's event queue for a few milliseconds. I'll post back with my findings...

AGlass0fMilk, 2019-08-17:
Still doesn't work -- my phone continuously attempts to connect and it goes into an infinite disconnection loop. This is seen with both the lightblue app and the nRFConnect app.

AGlass0fMilk, 2019-09-05:
Nearing 3 weeks...

AGlass0fMilk, 2019-09-05:
@pan @donatieng

maclobdell, 2019-09-05:
@bentcooke - any advice here?

AGlass0fMilk, 2019-09-05:
Just tried this with the nRF52840_DK from a fresh clone. Attempted to connect with my iPhone SE using the LightBlue scanner app. Still does not seem to be working properly. I get a connection/disconnection loop until the phone terminates the connection attempts with the following console output:

```
 PERIPHERAL 

f9:91:38:7f:c9:c8
Please connect to device
Connected to peer: 47:6e:c3:44:a1:3f
Error during SM::setLinkSecurity 4
Advertising timed out - aborting

 CENTRAL 

Diconnected

 PERIPHERAL 

f9:91:38:7f:c9:c8
Please connect to device
Connected to peer: 47:6e:c3:44:a1:3f
Error during SM::setLinkSecurity 4
Advertising timed out - aborting

 CENTRAL 

Diconnected                                                               
                                                                          
 PERIPHERAL                                                               
                                                                          
f9:91:38:7f:c9:c8                                                         
Please connect to device                                                  
Connected to peer: 47:6e:c3:44:a1:3f                                      
Error during SM::setLinkSecurity 4                                        
Advertising timed out - aborting                                          
                                                                          
 CENTRAL                                                                  
                                                                          
Diconnected                                                               
                                                                          
 PERIPHERAL                                                               
                                                                          
f9:91:38:7f:c9:c8                                                         
Please connect to device  
```

loverdeg-ep, 2019-09-06:
@0xc0170 @adbridge @donatieng @pan- @desmond-blue @paul-szczepanek-arm 

This has a customer behind it.
Further, there is potential for it and other products from this customer to connect to Pelion, via gateway or other.

0xc0170, 2019-09-09:
I'll talk to the team to check this issue.

40Grit, 2019-10-02:
@ranshe

simonpfeifhofer, 2019-10-08:
Any updates on this?

40Grit, 2019-11-01:
@0xc0170 This falls under the category of defect I believe

pan-, 2020-06-01:
@AGlass0fMilk Is it still an issue ? I can't reproduce it.

AGlass0fMilk, 2020-08-13:
This issue is now outdated and probably resolved thanks to updates to the BLE stack. If I encounter this again I will open a new issue.

closed 2020-08-13 by AGlass0fMilk