ARMmbed/mbed-os-example-ble #258: Periodic advertising example broken

lichangche, 2019-08-23:
The periodic adverting example breaks so far - in my test, the two nodes can establish their roles (after a number of changes including to change the first advertising type [not the periodic one] to legacy true, otherwise it spells out system error and crashes), the scanner will always return at this line:

    /* if we're looking for periodic advertising don't bother unless it's present */
    if (_role_established && !event.isPeriodicIntervalPresent()) {
        return;
    }
(From advertiser side, LE extended advs are happening but scanner can never detect the periodic interval !=0)

This has been raised in https://github.com/ARMmbed/mbed-os/issues/11106 however I think it is proper to create a separate issue as the original issue raised in 11106 has been resolved.

I've traced to the point where it shows the periodic interval field is always zero because when packets are constructed, it is always called isPeriodic false:

The same is true for all the instances where lctrPackAdvExtIndPdu are called.

pAdv->txAdvLen = lctrPackAdvExtIndPdu(pAdvSet, pAdvSet->advHdrBuf, FALSE);

Can someone please run the example and verify this?

lichangche, 2019-08-27:
Thanks for your attention.

Is there any update on this? Can anyone point me a Mbed OS version which works with this periodic advertising example so that I can continue with my experiment? - my target application is heavily rely on periodic advertising. 

Your kind help will be greatly appreciated.

pan-, 2020-10-05:
@lichangche We've updated recently the Bluetooth stack  used by Mbed OS. Does the problem persist ?

paul-szczepanek-arm, 2020-11-10:
Closing as this has been resolved on mbed-os and the current periodic advertising example works fine on nrf52. If you still see an issue please repoen @lichangche.

closed 2020-11-10 by paul-szczepanek-arm