ARMmbed/mbed-os-example-ble #259: No BLE examples work on MAX32630FTHR

idea--list, 2019-08-26:
Hi there,

I have the MAX32630FTHR board that has an onboard PAN1326B (that is based on TI's CC2564) dual-mode BT chip. I am having difficulties with running any of the BLE examples found [here](https://github.com/ARMmbed/mbed-os-example-ble) on this board. I am somewhat confused as this board is listed being BLE capable on the mbed website, the [github page](https://github.com/ARMmbed/mbed-os-example-ble) for the BLE examples tells one just needs a supported target.

How can i get the examples run? The documentation does not tell a word about this board.

idea--list, 2019-10-03:
Made some revision comparisons. Turns out that revision 72 (dated to 14 Dec 2018) of the examples work on this board even if one updates mbed-os to 5.14.

However later revisions of the examples do not work on this board. After comparing the main.cpp files it turns out the skeleton of the examples was rewritten, so that is the cause why all the examples are broken for this board. All that means the issue is independent from mbed-os, and main.cpp of the examples might need to be reviewed.

pan-, 2020-10-01:
@idea--list This board doesn't receive active support from it's manufacturer and is not part of mbed OS 6. 
You can report the problem back to Maxim.

closed 2020-10-01 by pan-