ARMmbed/mbed-os-example-ble #261: BLE_GapButton example doesn't work

pan-, 2019-09-05:
I've noticed several issues on the example:

The advertising payload isn't updated into its advertising set; the following lines are missing in the function [update_button_payload](https://github.com/ARMmbed/mbed-os-example-ble/blob/47a653482689be385512aa522570c343b8471843/BLE_GAPButton/source/main.cpp#L136)

https://github.com/ARMmbed/mbed-os-example-ble/blob/47a653482689be385512aa522570c343b8471843/BLE_GAPButton/source/main.cpp#L116-L119

Event with this change update doesn't occur on `DISCO_L475VG_IOT01A` . However it works on `NRF52840_DK`. 

On `NRF52840` the button doesn't work out of the box. 

Finally, the field [`_button_count`](https://github.com/ARMmbed/mbed-os-example-ble/blob/47a653482689be385512aa522570c343b8471843/BLE_GAPButton/source/main.cpp#L179) is not initialised properly. 

@desmond-blue @donatieng This is blocking @ashok-rao .

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

pan-, 2020-10-02:
Similar to #215

paul-szczepanek-arm, 2020-10-21:
This example is being replaced. The new example will not have this issue.

closed 2020-10-23 by ciarmcom