ARMmbed/mbed-os-example-ble #262: BLE DTM Test

AGlass0fMilk, 2019-09-05:
We are nearing production with a BLE device and will need to perform Bluetooth Qualification.

Is there any example of BLE DTM based on Mbed's Pelion stack?

pan-, 2019-09-05:
@donatieng

AGlass0fMilk, 2019-09-05:
This may be more hardware specific but there should be a HAL/API to port DTM commands and an example application using Mbed's UART interface. Otherwise, we'll have to rely on Nordic's SDK for this.

maclobdell, 2019-09-19:
@donatieng - Please let us know if you have any guidance.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

paul-szczepanek-arm, 2020-11-10:
We don't have an example but the test functionality in cordio is now in the CordioHCIDriver (rf_test_start_le_receiver_test, rf_test_start_le_transmitter_test, rf_test_end). We don't have plans of making an example for the usage but I'd welcome a PR for one.

closed 2020-11-10 by paul-szczepanek-arm