ARMmbed/mbed-os-example-ble #296: BLE Gatt Client (and others?) doesn't build for NRF52840_DK on Mbed OS 5.15.3

ifyall, 2020-05-13:
The BLE Gatt Client example is failing at the link step due to functions being missing in the final image. See the error messages below.

Mbed OS version: 5.15.3
mbed-os-example-ble SHA: 03eb4a5e

[Error] pretty_printer.h@83,29: no matching function for call to 'ble::own_address_type_t::own_address_type_t()'
[Error] pretty_printer.h@85,56: no matching function for call to 'ble::interface::LegacyGap<ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> > >::getAddress(ble::own_address_type_t&, ble::address_t&)'
[Warning] AdvertisingDataParser.h@69,22: comparison between signed and unsigned integer expressions [-Wsign-compare]
[Warning] AdvertisingDataParser.h@78,41: comparison between signed and unsigned integer expressions [-Wsign-compare]
[ERROR] In file included from .\source\BLEProcess.h:21:0,
                 from .\source\main.cpp:30:
.\source\pretty_printer.h: In function 'void print_mac_address()':
.\source\pretty_printer.h:83:29: error: no matching function for call to 'ble::own_address_type_t::own_address_type_t()'
     ble::own_address_type_t addr_type;
                             ^~~~~~~~~
In file included from ./mbed-os/features/FEATURE_BLE/ble/BLETypes.h:25:0,
                 from ./mbed-os/features/FEATURE_BLE/ble/GattAttribute.h:21,
                 from ./mbed-os/features/FEATURE_BLE/ble/GattClient.h:21,
                 from .\source\main.cpp:25:
./mbed-os/features/FEATURE_BLE/ble/gap/Types.h:578:5: note: candidate: ble::own_address_type_t::own_address_type_t(ble::own_address_type_t::type)
     own_address_type_t(type value) : SafeEnum(value)
     ^~~~~~~~~~~~~~~~~~
./mbed-os/features/FEATURE_BLE/ble/gap/Types.h:578:5: note:   candidate expects 1 argument, 0 provided
./mbed-os/features/FEATURE_BLE/ble/gap/Types.h:546:8: note: candidate: constexpr ble::own_address_type_t::own_address_type_t(const ble::own_address_type_t&)
 struct own_address_type_t : SafeEnum<own_address_type_t, uint8_t> {
        ^~~~~~~~~~~~~~~~~~
./mbed-os/features/FEATURE_BLE/ble/gap/Types.h:546:8: note:   candidate expects 1 argument, 0 provided
./mbed-os/features/FEATURE_BLE/ble/gap/Types.h:546:8: note: candidate: constexpr ble::own_address_type_t::own_address_type_t(ble::own_address_type_t&&)
./mbed-os/features/FEATURE_BLE/ble/gap/Types.h:546:8: note:   candidate expects 1 argument, 0 provided
In file included from .\source\BLEProcess.h:21:0,
                 from .\source\main.cpp:30:
.\source\pretty_printer.h:85:56: error: no matching function for call to 'ble::interface::LegacyGap<ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> > >::getAddress(ble::own_address_type_t&, ble::address_t&)'
     BLE::Instance().gap().getAddress(addr_type, address);
                                                        ^
In file included from ./mbed-os/features/FEATURE_BLE/ble/ServiceDiscovery.h:21:0,
                 from ./mbed-os/features/FEATURE_BLE/ble/GattClient.h:22,
                 from .\source\main.cpp:25:
./mbed-os/features/FEATURE_BLE/ble/Gap.h:835:17: note: candidate: ble_error_t ble::interface::LegacyGap<Impl>::getAddress(BLEProtocol::AddressType_t*, uint8_t*) [with Impl = ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >; BLEProtocol::AddressType_t = BLEProtocol::AddressType::Type; uint8_t = unsigned char]
     ble_error_t getAddress(
                 ^~~~~~~~~~
./mbed-os/features/FEATURE_BLE/ble/Gap.h:835:17: note:   no known conversion for argument 1 from 'ble::own_address_type_t' to 'BLEProtocol::AddressType_t* {aka BLEProtocol::AddressType::Type*}'
In file included from .\source\BLEProcess.h:29:0,
                 from .\source\main.cpp:30:
./mbed-os/features/FEATURE_BLE/ble/gap/AdvertisingDataParser.h: In member function 'bool ble::AdvertisingDataParser::hasNext() const':
./mbed-os/features/FEATURE_BLE/ble/gap/AdvertisingDataParser.h:69:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (position >= data.size()) {
             ~~~~~~~~~^~~~~~~~~~~~~~
./mbed-os/features/FEATURE_BLE/ble/gap/AdvertisingDataParser.h:78:41: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (position + current_length() >= data.size()) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~

[mbed] Working path "C:\Users\fya\mbed\mbed-os-example-ble\BLE_GattClient" (library)
[mbed] Program path "C:\Users\fya\mbed\mbed-os-example-ble\BLE_GattClient"
[mbed] ERROR: "c:\users\fya\appdata\local\programs\python\python37\python.exe" returned error.
       Code: 1
       Path: "C:\Users\fya\mbed\mbed-os-example-ble\BLE_GattClient"
       Command: "c:\users\fya\appdata\local\programs\python\python37\python.exe -u C:\Users\fya\mbed\mbed-os-example-ble\BLE_GattClient\mbed-os\tools\make.py -t GCC_ARM -m NRF52840_DK --source . --build .\BUILD\NRF52840_DK\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output

pan-, 2020-10-01:
@ifyall Is it still an issue ? What version of the examples where you using and how did you compiled them ?

paul-szczepanek-arm, 2020-11-10:
Since the examples have been overhauled since, I'm closing this. If you're still having issues with the current version @ifyall please reopen.

closed 2020-11-10 by paul-szczepanek-arm