ARMmbed/mbed-os-example-ble #308: Why are none of the examples using the UUID class to properly define UUIDs

Gerriko, 2020-06-26:
I'm looking at the MBED ble examples for the first time and I am rather perplexed as to why none of the examples provided on GitHub use the UUID class to define the UUID's.

Otherwise, what is the purpose of UUID.h?

I'm ignoring the fact that some of these examples, like LEDservice, are also defining the services/ characteristics as 16 bit short UUID's, which not considered best practice, IMHO.

Gerriko, 2020-06-27:
In hindsight, I should've broadened this issue.

I am finding that there is very little commonality of approach in any of the examples provided. As these are official MBED examples, it would be really helpful to the new user if things could be tidied up a bit, as it takes ages trawling through each code example to make sense of all the different approaches and coding styles and try to work out if the approach take is specific to the example functionality or merely because that this the way the coder approached it. Do the coders not review and try and apply a common approach? In my view, it would be helpful to split GAP functionality in one class and GATT functionality as another and then create specific files for each so that they can be reused in each of the examples, if applicable.

For example, there is the BLE_GattServer example, which correctly uses 128-bit UUID's but then has the Gatt Service code within main.cpp, while the BLE_LED example has used 16-bit UUID's and has the LED GATT service in a separate .h file, for some arbitrary reason. Then the BLE_GattServer example has extracted out a wrapper class called BLEProcess as a separate .h file but this approach is not replicated in the BLE_LED example, when it could have. This makes things very confusing to the reader and slows the learning process down (in my opinion).

pan-, 2020-11-09:
@Gerriko If you look at the development branch, significant improvements have been made to the example to make them more consistent.  Use of 128bits UUID is demonstrated in https://github.com/ARMmbed/mbed-os-example-ble/blob/development/BLE_GattServer_CharacteristicUpdates/source/main.cpp . 

Do you have other suggestions ?

closed 2020-11-19 by pan-