ARMmbed/mbed-os-example-ble #315: BLE examples throwing up errors in MBED Studio (OS6.2)

Gerriko, 2020-08-06:
I tested some of these examples over a month ago and they compiled (I believe it was OS 6.1).

Now, I have just tried to compile two examples on MBED Studio (OS6.2) and these threw up some "generic" errors for the main.cpp and in the "pretty_printer.h" file.

In file included from ./mbed-os/features/FEATURE_BLE/ble/BLE.h:20:
./mbed-os/features/FEATURE_BLE/ble/BLERoles.h:5:6: error: "BLE requires at least one role 'BROADCASTER' or 'OBSERVER' to be enabled"
#error "BLE requires at least one role 'BROADCASTER' or 'OBSERVER' to be enabled"
^
In file included from ./main.cpp:19:
In file included from ./mbed-os/features/FEATURE_BLE/ble/BLE.h:23:
In file included from ./mbed-os/features/FEATURE_BLE/ble/Gap.h:20:
./mbed-os/features/FEATURE_BLE/ble/gap/Gap.h:20:10: fatal error: 'BLERoles.h' file not found
#include "BLERoles.h"

Then for pretty_printer.h
Unknown type name 'ble_error_t'; did you mean 'mbed::ble_error_t'?
Use of undeclared identifier 'BLE_ERROR_NONE'; did you mean 'mbed::BLE_ERROR_NONE'?
... and this is repeated for all error cases in switch statement.
Type 'const mbed::ble::address_t' does not provide a subscript operator

Gerriko, 2020-08-06:
I have just discovered that if I use "**mbed-os-empty**" as my template or starting point then it does compile without error, although I am getting warnings because of the use of deprecated functions in main.cpp. 

For example MBED Studio informs me that "call_every" is deprecated since MBED OS 6.0 and we are now at MBED OS 6.2:
`_blink_event = _event_queue.call_every(500, this, &GapDemo::blink);`

The problem highlighted above arose when I used the "**mbed-os-example-blinky-baremetal**" as my starting point.

pan-, 2020-10-21:
@Gerriko https://github.com/ARMmbed/mbed-os-example-blinky-baremetal contains a list of required dependencies in its [mbed_app.json](https://github.com/ARMmbed/mbed-os-example-blinky-baremetal/blob/master/mbed_app.json). Did you had the BLE dependencies to it ? If not the compilation errors seems normal but that doesn't explained why it worked with Mbed OS 6.1.  

I don't know how mbed studio deals with these compilation requirement but that seems to be an issue to raise to the mbed studio team on mbed-os forums and not in this example repository.  

The point on warnings due to call to deprecated function is being addressed by https://github.com/ARMmbed/mbed-os-example-ble/pull/328 .

closed 2020-10-21 by pan-