ARMmbed/mbed-os-example-ble #319: BLE_GattServer does not appear to advertise its service UUID

Gerriko, 2020-08-16:
I noticed when testing the BLE_GattServer example with a BLE utility app, like nRFConnect or Bluetooth LE Scanner, that the **service UUID is not advertised**.

It is more common than not for a service UUID to be advertised.

Gerriko, 2020-08-17:
Thanks. It appears that within the function `void start_advertising()` ALL advertising parameters have not been defined as this is empty `ble::AdvertisingParameters adv_params;`

pan-, 2020-10-01:
@Gerriko The application is supposed to set in the advertising payload the services UUIDs it wants to advertise.

closed 2020-10-07 by pan-

Gerriko, 2020-10-07:
@pan- Not sure why this is closed as whilst you are correct that an application is supposed to set up payload and service UUID, as far as I could make out, it is currently NOT possible to define and then advertise a 128 bit service UUID with the Mbed BLE library. You can readily do it using a 16bit UUID but not 128bit. Maybe this can be clarified, otherwise I plan to reopen the issue. Thank you.

pan-, 2020-10-07:
@Gerriko 

The goal of that example is to demonstrate basic functionalities of the GattServer API. That is the reason why advertising is not seen in the main file we don't want to expose too many unrelated details in the example.

However I admit it would be valuable for the community to have a blueprint of what is possible with BLE API from A to Z.    

To set local service UUID in the advertising payload you can use the function `AdvertisingDataBuilder::setLocalServiceList`.  It is [here](https://github.com/ARMmbed/mbed-os/blob/ad40b1b2672d59b058f7c104fdc612d0dddf3ac3/connectivity/FEATURE_BLE/include/ble/gap/AdvertisingDataBuilder.h#L289-L302). 

If you have issues with this function, I encourage you to open an issue in mbed OS.