ARMmbed/mbed-os-example-ble #335: Why is delay required before starting advertisements after setting up SecurityManager?

AGlass0fMilk, 2020-10-29:
I am adding pairing/privacy to a new BLE application and was following the `BLE_SM` example in this repo. After adding the necessary changes in, my BLE device was no longer advertising. After checking the debug UART output, I found that for some reason the call to `Gap::startAdvertising()` was failing!

The fix ended up being to add a small delay (using `EventQueue::call_in` instead of simply `call`) to begin advertising after setting up the security manager, as is done in the example:

https://github.com/ARMmbed/mbed-os-example-ble/blob/a43debcf2ab49b18aa8e99c7437ca5d56aafe653/BLE_SM/source/main.cpp#L185-L186

Why is this delay necessary? Does it have something to do with the filesystem?

pan-, 2020-10-29:
Privacy takes some time to setup. During that time, it is not possible to advertise otherwise you'd advertise with a non private address. The API signal to the user through a callback when privacy has been enabled and gap is ready to be used (advertising, scan, connection).

AGlass0fMilk, 2020-10-29:
Ah, that's quite a recent update:

https://github.com/ARMmbed/mbed-os/commit/bb35cba134fb6d9a09c6391e39d1946e61486929

I will have to update `ChainableGapEventHandler` to support the new `onPrivacyEnabled()` callback.

Perhaps the example should be updated in the future to explain this?

AGlass0fMilk, 2020-10-29:
@pan- I added changes to address this question in the future into my BLE_SM PR. See above linked issue.

pan-, 2020-10-29:
Thanks @AGlass0fMilk I believe changes where lost because it was present in https://github.com/ARMmbed/mbed-os-example-ble/pull/323/files . I missed it during the review of the example rework.

paul-szczepanek-arm, 2020-10-29:
I have included that in the rework of examples. Unfortunately due to unfortunate timing I couldn't merge my PR until after release. Please look at the development branch of examples.

AGlass0fMilk, 2020-10-30:
Question has been answered and resolution is currently in the development branch, it will be in master upon next release.

closed 2020-10-30 by AGlass0fMilk