ARMmbed/mbed-os-example-ble #339: BLE_HeartRate example does not work on NRF52_DK

jw-smaal, 2020-11-08:
### Description of defect

BLE_HearRate example compiles ok.  However when using a BLE scanner and then connecting to the "Heartrate" device it disconnects unexpectedly.   I tried multiple types of BLE scanners same issue on iPhone. 
  
The DAPLINK CMSIS-DAP debug printf only printouts: DEVICE MAC ADDRESS: da:3f:a1:a:f5:f 
 
#### Target(s) affected by this defect ?

NRF_52_DK 
#### Toolchain(s) (name and version) displaying this defect ?

MBED-CLI + GCC
Building project BLE_HeartRate (NRF52_DK, GCC_ARM)
Scan: BLE_HeartRate
Link: BLE_HeartRate
Elf2Bin: BLE_HeartRate
| Module               |      .text |    .data |      .bss |
|----------------------|------------|----------|-----------|
| [fill]               |    512(+0) |   10(+0) |    82(+0) |
| [lib]/c.a            |   7732(+0) | 2112(+0) |    56(+0) |
| [lib]/gcc.a          |    920(+0) |    0(+0) |     0(+0) |
| [lib]/misc           |    240(+0) |    4(+0) |    28(+0) |
| [lib]/stdc++.a       |      0(+0) |    0(+0) |     0(+0) |
| mbed-os/cmsis        |   7808(+0) |  168(+0) |  5956(+0) |
| mbed-os/connectivity | 205748(+0) |  286(+0) | 23424(+0) |
| mbed-os/drivers      |    798(+0) |    0(+0) |     0(+0) |
| mbed-os/events       |   1250(+0) |    0(+0) |     0(+0) |
| mbed-os/hal          |   1406(+0) |    8(+0) |   129(+0) |
| mbed-os/platform     |   5400(+0) |  260(+0) |   416(+0) |
| mbed-os/rtos         |    184(+0) |    0(+0) |     8(+0) |
| mbed-os/targets      |   9448(+0) |    8(+0) |   681(+0) |
| source/main.o        |   1578(+0) |    0(+0) |    92(+0) |
| Subtotals            | 243024(+0) | 2856(+0) | 30872(+0) |
Total Static RAM memory (data + bss): 33728(+0) bytes
Total Flash memory (text + data): 245880(+0) bytes

Image: ./BUILD/NRF52_DK/GCC_ARM/BLE_HeartRate.hex

#### What version of Mbed-os are you using (tag or sha) ?
https://github.com/ARMmbed/mbed-os/#8ef0a435b2356f8159dea8e427b2935d177309f8

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed cli 1.10.4 
arm-none-eabi-gcc-9.2.1 

#### How is this defect reproduced ?
simply try to compile to the example with a NRF52_DK target and connect to it using a BLE scanner.  I also tried other examples (Battery) and gives the same issue).

paul-szczepanek-arm, 2020-11-09:
Can you look into the log on your iphone to see what the reason for disconnection is? You can also add to the example after init complete:
```
        _gap.setEventHandler(this);
```
and add a function to the main class:
```
    void onDisconnectionComplete(const ble::DisconnectionCompleteEvent &event) override
    {
        printf("Disconnected with error %d\r\n", event.getReason());
    }
```
This way the example will also print the reason for disconnection.

jw-smaal, 2020-11-09:
Paul,

Thank you for looking into this.  The iPhone BLE scanner reports:
_Connected --> Discovering Services -- > [Callback] centralManager(centraldidDisconnectPeripheral: peripheral error: "The connection has times out unexpectedly"_  

When I add the code I get 
DEVICE MAC ADDRESS: da:3f:a1:a:f5:f
**Disconnected with error 8**

paul-szczepanek-arm, 2020-11-09:
That means a timeout, one of the devices failed to send a packet for longer than the supervision timeout.

It's worthwhile to check the connection parameters. If you don't see them on your phone you can do a similar event on the example:
void onConnectionComplete(const ble::ConnectionCompleteEvent &event) override

and examine:
        conn_interval_t connectionInterval,
        slave_latency_t connectionLatency,
        supervision_timeout_t supervisionTimeout,
If for some reason your phone is busy with other traffic it might miss connection slots or maybe you have a noisy environment and despite the board connecting the quality of the connection is so poor that it barely ever manages to get a packet through and disconnects. Check with your scanner to see the dbm of the signal and how many devices are competing for the spectrum.

paul-szczepanek-arm, 2020-11-09:
This is not related to your problem (at least I don't) but wanted to mention that I have recently overhauled the examples - the new versions are available in the 'development' branch.

jw-smaal, 2020-11-09:
I have a good signal -37dBm I tried the development branch as well and with the BLE_GAP example I am getting similar issues.  Maybe it's not a bug but I am unable to get it to function at least with my phone.  I am going to get some other devices to try with as well.  

The serial port on the NRF52_DK shows the following:

> 
> We have scanned for 200ms with an interval of 80 timeslots and a window of 60 timeslots
> We have been listening on the radio for at least 150ms
> 
> Advertising started (type: 0x0, interval: [25 : 50]ms)
> Gap::createAdvertisingSet() failed: BLE_ERROR_NO_MEM: Out of Memory
> Advertising set 0 started
> Connected in 200ms
> Stopped advertising early due to connection
> Data length changed on the connection 1.
> Maximum sizes for over the air packets are:
> 251 octets for transmit and 251 octets for receive.
> Phy update on connection 1 - Tx Phy: LE 2M, Rx Phy: LE 2M
> Data length changed on the connection 1.
> Maximum sizes for over the air packets are:
> 251 octets for transmit and 251 octets for receive.
> Disconnected
> We have advertised for 200ms
> We created at least 4 tx and rx events
> 
> Scanning started (interval: 50ms, window: 37ms, timeout: 10000ms).
> We found a connectable device
> Connected in 349ms
> Disconnected
> We have scanned for 349ms with an interval of 80 timeslots and a window of 60 timeslots
> We have been listening on the radio for at least 225ms
> 
> Advertising started (type: 0x0, interval: [25 : 50]ms)
> Gap::createAdvertisingSet() failed: BLE_ERROR_NO_MEM: Out of Memory
> Advertising set 0 started

paul-szczepanek-arm, 2020-11-10:
The BLE_GAP example is designed to connect and disconnect on its own. It looks like it's happily negotiating higher bandwidth and longer packet size so your connection must be OK. I'm surprised by the out of memory errors for creation of an advertising set. I'll look into that.
On of the examples designed to maintain a connection is any of the gatt ones like the BLE_GattServer_CharacteristicWrite which will wait for connection and allow you to write a characteristic at your leisure.

jw-smaal, 2020-11-10:
@paul-szczepanek-arm happy to report that I managed to use the "BLE_GattServer_CharacteristicWrite" connect to the GattServer on the NRF52_DK and write characteristic to it.   Thanks for all your help.

paul-szczepanek-arm, 2020-11-11:
Glad it works for you. I will follow up on the out of memory errors in the gap example but since it's not part of this issue I will close this. If you think there's still a problem please reopen.

closed 2020-11-11 by paul-szczepanek-arm